graugans/meta-udoo #2: Support for UDOO Quad/Dual power button

graugans, 2016-02-11:
The Quad/Dual boards are equipped with a PWR button. when shutting down the board with 
```bash
$ halt
```
The board wakes up again but during runtime nothing happens when pressing this button.

Some details from the [i.mx6 hardware dev guide](http://cache.freescale.com/files/32bit/doc/user_guide/IMX6DQ6SDLHDG.pdf)
[image]

There is a driver for [snvs-pwrkey](https://github.com/UDOOboard/linux_kernel/blob/23c08ac885a2185bd12af9696653d47f587eddc7/Documentation/devicetree/bindings/input/snvs-pwrkey.txt) but seems not activated by device tree.

graugans, 2016-05-17:
Here is a workaround for the issue: http://www.clarenceho.net/2016/05/adding-shutdown-button-to-udoo.html?m=1

modjo756, 2016-11-24:
I think to add the pwr button working, need to add this during the kernel build : 
[power](https://secogit.seco.com/imx6_release/linux-3-14-28-secoboards-imx6_rel/commit/fff7e6853a574187a38dfb3e89f18efd83c446ec#a09b3da08a589c8d1fa12bff24024269c0cdf6dc_3352_3352)