graugans/meta-udoo #3: M4 firmware

modjo756, 2016-02-11:
The file m4last.fw is ,ot created on the folder /var/opt/M4 after the boot. Need to put it manually.

graugans, 2016-02-11:
Strange, for me it is working after a fresh build and uploading to UDOONeo

```bash
root@udooneo:~# ls /var/opt/m4/m4last.fw 
/var/opt/m4/m4last.fw
```

modjo756, 2016-02-11:
Ok, this week end i will try to re-build yocto and load the card another time.

graugans, 2016-02-12:
Please do not reflash. You can check if in the init script (``/etc/init.d/udoofota``) there is the following line:
```
 mkdir -p /var/opt/m4/
```
In case it is still not working as expected you can run 
```` bash
$ /etc/init.d/udoofota stop
$ udooserver
```
And check the output

modjo756, 2016-02-13:
Hi graugans, i rebuild a clean yocto for neo (i just saw your message this morning !) but same problem after a boot, see after a log under ssh(tty, eth not working, i will open an issue) : 

`root@udooneo:~# cat /etc/init.d/udoofota 
#!/bin/sh
#
#   UDOO Firmware Over The Air Server
#

# Script variables
PROG=udooserver
PID=/var/run/udoofota-server.pid
RETURN_VAL=0

start()
{
    # create m4 upload folder
    mkdir -p /var/opt/m4/
    echo "[`date -u '+%F %T %Z'`] (sys) Starting" >> /var/log/udoofota-server.log
    test -x /usr/bin/udooneo-m4uploader ||
    {
        stop
        echo "[`date -u '+%F %T %Z'`] (sys) Error: uploader not found" >> /var/log/udoofota-server.log
        exit 0
    }

    echo "Starting $PROG: "
    start-stop-daemon \
    --start \
    --pidfile $PID \
    -b --exec /usr/bin/udooserver >> /var/log/udoofota-server.log 2>&1
    return $?
}

stop()
{
    echo "[`date -u '+%Y-%m-%d %T %Z'`] (sys) Stopping" >> /var/log/udoofota-server.log
    start-stop-daemon --stop --quiet --pidfile $PID
    return $?
}

status() {
        if [ -s $PID ]; then
                echo "$PROG is running as pid `cat $PID`:"
        else
                echo "$PROG is not running."
        fi
}

restart()
{
    stop
    start
    return $?
}

case "$1" in
 start)
       start
    ;;
 stop)
       stop
    ;;
 status)
       status
    ;;
 restart|reload)
       restart
    ;;
 *)
    /bin/echo "Usage: $0 {start|stop|status|restart|reload}"
    RETURN_VAL=1
esac

exit $RETURN_VAL
root@udooneo:~# cat /var/opt/m4/
cat: /var/opt/m4/: Is a directory
root@udooneo:~# ls /var/opt/m4 
root@udooneo:~# /etc/init.d/udoofota stop
root@udooneo:~# udooserver
bind: Address already in use
root@udooneo:~# 
`

modjo756, 2016-03-06:
Graugans, it seems that now it working correctly (perhaps i'm wrong before) i reload two times a sd card and i can load from arduino ide the board --> you can close the bug (m4last.fw is present on folder)

graugans, 2016-03-07:
I will close this after a reorganization. The details are described [here](https://github.com/gmacario/easy-jenkins/issues/62)

commit 31a74069 (2016-03-07): Added m4-firmware to MACHINE_ESSENTIAL_EXTRA_RDEPENDS

So the m4-firmware is also available to the core-image-minimal image.
Otherwise it is not included in images not including  packagegroup-base

This fixes #3
closed 2016-03-07 by graugans