graugans/meta-udoo #5: problem with ssh access

modjo756, 2016-02-13:
sometimes on neo (after burning a new image) ssh over eth not workink. test with eth0 up, ping work, under a ps when ssh over eth not working : 

`root@udooneo:~# ps
  PID TTY          TIME CMD
  467 ttyGS0   00:00:00 sh
  481 ttyGS0   00:00:00 ps
root@udooneo:~# 
`

graugans, 2016-02-13:
Can you please run 

```bash
$ ps aux
```

modjo756, 2016-02-13:
i have reburn a sdcard and now works (i do this when i meet this issue), result of ps aux (with working ssh) : 

`USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.5  0.0   1716   584 ?        Ss   23:46   0:01 init [5]
root         2  0.0  0.0      0     0 ?        S    23:46   0:00 [kthreadd]
root         3  0.0  0.0      0     0 ?        S    23:46   0:00 [ksoftirqd/0]
root         4  0.0  0.0      0     0 ?        S    23:46   0:00 [kworker/0:0]
root         5  0.0  0.0      0     0 ?        S<   23:46   0:00 [kworker/0:0H]
root         6  0.0  0.0      0     0 ?        S    23:46   0:00 [kworker/u2:0]
root         7  0.0  0.0      0     0 ?        S    23:46   0:00 [rcu_preempt]
root         8  0.0  0.0      0     0 ?        S    23:46   0:00 [rcu_sched]
root         9  0.0  0.0      0     0 ?        S    23:46   0:00 [rcu_bh]
root        10  0.0  0.0      0     0 ?        S    23:46   0:00 [migration/0]
root        11  0.0  0.0      0     0 ?        S<   23:46   0:00 [khelper]
root        12  0.0  0.0      0     0 ?        S    23:46   0:00 [kdevtmpfs]
root        13  0.0  0.0      0     0 ?        S<   23:46   0:00 [writeback]
root        14  0.0  0.0      0     0 ?        S<   23:46   0:00 [bioset]
root        15  0.0  0.0      0     0 ?        S<   23:46   0:00 [crypto]
root        16  0.0  0.0      0     0 ?        S<   23:46   0:00 [kblockd]
root        17  0.0  0.0      0     0 ?        S    23:46   0:00 [kworker/u2:1]
root        28  0.0  0.0      0     0 ?        S    23:46   0:00 [cfinteractive]
root        29  0.1  0.0      0     0 ?        S    23:46   0:00 [kworker/0:1]
root        30  0.0  0.0      0     0 ?        S    23:46   0:00 [kswapd0]
root        31  0.0  0.0      0     0 ?        S    23:46   0:00 [fsnotify_mark]
root        68  0.0  0.0      0     0 ?        S    23:46   0:00 [pxp_dispatch]
root        71  0.0  0.0      0     0 ?        S    23:46   0:00 [irq/324-st1232]
root        72  0.0  0.0      0     0 ?        S    23:46   0:00 [irq/322-2194000]
root        73  0.1  0.0      0     0 ?        S    23:46   0:00 [mmcqd/0]
root        74  0.0  0.0      0     0 ?        S<   23:46   0:00 [deferwq]
root        75  0.0  0.0      0     0 ?        S    23:46   0:00 [irq/81-imx_ther]
root        76  0.0  0.0      0     0 ?        S<   23:46   0:00 [kworker/0:1H]
root        77  0.0  0.0      0     0 ?        S    23:46   0:00 [jbd2/mmcblk0p2-]
root        78  0.0  0.0      0     0 ?        S<   23:46   0:00 [ext4-rsv-conver]
root       110  0.0  0.1   3548  1084 ?        Ss   23:46   0:00 /lib/udev/udevd -d
root       124  0.0  0.0      0     0 ?        S<   23:46   0:00 [galcore workque]
root       125  0.0  0.0      0     0 ?        S    23:46   0:00 [khubd]
root       130  0.0  0.0      0     0 ?        S<   23:46   0:00 [ci_otg]
root       142  0.0  0.0   3564   768 ?        S    23:46   0:00 /lib/udev/udevd -d
root       143  0.0  0.0   3564   776 ?        S    23:46   0:00 /lib/udev/udevd -d
root       146  0.0  0.0      0     0 ?        S<   23:46   0:00 [cfg80211]
root       153  0.0  0.0      0     0 ?        S    23:46   0:00 [galcore daemon ]
root       154  0.0  0.0      0     0 ?        S<   23:46   0:00 [wl12xx_wq]
root       155  0.0  0.0      0     0 ?        S    23:46   0:00 [kworker/0:2]
root       175  0.0  0.0      0     0 ?        S    23:46   0:00 [irq/208-wl18xx]
root       317  0.0  0.0   2724   252 ?        Ss   23:46   0:00 udhcpc -R -b -p /var/run/udhcpc.eth0.pid -i eth0
message+   324  0.0  0.0   3012   784 ?        Ss   23:46   0:00 /usr/bin/dbus-daemon --system
root       332  0.2  0.0      0     0 ?        S<   23:46   0:00 [kworker/u3:0]
root       333  0.0  0.0      0     0 ?        S<   23:46   0:00 [hci0]
root       334  0.0  0.0      0     0 ?        S<   23:46   0:00 [hci0]
root       335  0.0  0.0   1568   248 ?        S    23:46   0:00 /usr/sbin/uim -f /sys/./devices/soc0/kim.22
root       337  0.0  0.0      0     0 ?        S<   23:46   0:00 [kworker/u3:1]
root       345  0.0  0.0      0     0 ?        S<   23:46   0:00 [ipv6_addrconf]
root       346  0.0  0.0   4908   740 ?        Ss   23:46   0:00 /usr/sbin/sshd
rpc        350  0.0  0.0   2344   756 ?        Ss   23:46   0:00 /usr/sbin/rpcbind
rpcuser    355  0.0  0.0   2136   712 ?        Ss   23:46   0:00 /usr/sbin/rpc.statd
root       363  0.0  0.0   2176   608 ?        S    23:46   0:00 /usr/sbin/atd -f
root       368  0.0  0.0      0     0 ?        S<   23:46   0:00 [rpciod]
root       374  0.0  0.0      0     0 ?        S<   23:46   0:00 [nfsd4]
root       375  0.0  0.0      0     0 ?        S<   23:46   0:00 [nfsd4_callbacks]
root       376  0.0  0.0      0     0 ?        S    23:46   0:00 [lockd]
root       379  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       380  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       381  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       382  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       383  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       384  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       385  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       386  0.0  0.0      0     0 ?        S    23:46   0:00 [nfsd]
root       388  0.0  0.0   2660   492 ?        Ss   23:46   0:00 /usr/sbin/rpc.mountd -f /etc/exports 
[... cut ...]
root       431  0.0  0.0   2728   636 ttymxc0  Ss+  23:46   0:00 /sbin/getty -L 115200 ttymxc0
root       432  0.0  0.0   2728   636 ttyGS0   Ss+  23:46   0:00 /sbin/getty -L 115200 ttyGS0
root       433  0.0  0.0   2728   636 tty1     Ss+  23:46   0:00 /sbin/getty 38400 tty1
root       434  0.0  0.0      0     0 ?        S<   23:46   0:00 [kworker/u3:2]
root       442  2.0  0.2   4996  2148 ?        Ss   23:50   0:00 sshd: root@pts/0
root       444  0.5  0.1   3244  1448 pts/0    Ss   23:50   0:00 -sh
root       445  0.0  0.0   3040   980 pts/0    R+   23:50   0:00 ps aux
`

closed 2016-04-04 by modjo756