graugans/meta-udoo #6: lvds 7 inchs (neo)

modjo756, 2016-02-13:
Try to launch neo with lvds7 inchs, configure uEnv.txt to use lvds in place oh HDMI but problem during the boot, see the log : 

`U-Boot SPL 2015.04 (Feb 12 2016 - 22:48:09)
Setting 1024MB RAM calibration data
port 1

U-Boot 2015.04 (Feb 12 2016 - 22:48:09)

CPU:   Freescale i.MX6SX rev1.2 at 792 MHz
CPU:   Temperature 51 C
Reset cause: POR
Board: UDOO Neo Full
I2C:   ready
DRAM:  1 GiB
PMIC:  PFUZE3000 DEV_ID=0x30 REV_ID=0x11
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   CPU Net Initialization Failed
No ethernet found.
Normal Boot
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
reading uEnv.txt
164 bytes read in 10 ms (15.6 KiB/s)
Running bootscript from mmc ...
Device Tree: imx6sx-udoo-neo-full-lvds7-m4.dtb
107976 bytes read in 192 ms (548.8 KiB/s)
## Starting auxiliary core at 0x84000000 ...
reading /zImage
4269240 bytes read in 211 ms (19.3 MiB/s)
Booting from mmc ...
reading imx6sx-udoo-neo-full-lvds7-m4.dtb
** Unable to read file imx6sx-udoo-neo-full-lvds7-m4.dtb **
Kernel image @ 0x80800000 [ 0x000000 - 0x4124b8 ]
No gpc device node -9, force to ldo-enable.

Starting kernel ...
`
The file imx6sx-udoo-neo-full-lvds7-m4.dtb is not present in boot folder, copy it from udoobuntu2 and it works perfect, just the touch don't work under qtapp, not tested if works under linux

closed 2016-02-13 by graugans

graugans, 2016-02-14:
Can you please post your changes to uEnv.txt and open a new issue for touch?

modjo756, 2016-02-14:
my change to uEnv.txt

#save this file with unix-like newlines! (\n, 0x0A)

video_output=lvds7
m4_enabled=true

commit f3705120 (2016-02-13): Added missing device tree for 7" LVDS + M4 screen

This closes #6.