graugans/meta-udoo #9: touch lvds7 (neo and quad) not working in Qt app

modjo756, 2016-02-14:
when i launch a Qt app on neo or quad, the touch is not working (i added tslib in my local.conf). Following this link to test the touch : [yocto touch](http://www.lindusembedded.com/blog/2013/02/07/adding-touch-support-in-yocto/).
under the result : 

```
$ dmesg (quad)
 
input: st1232-touchscreen as /devices/soc0/soc.0/2100000.aips-bus/21a8000.i2c/i2c-2/2-0055/input/input0

$ dmesg (neo)

input: st1232-touchscreen as /devices/soc0/soc.0/2100000.aips-bus/21a0000.i2c/i2c-0/0-0055/input/input1
[    1.154162] i2c /dev entries driver
[    1.158368] IR NEC protocol handler initialized
[    1.162915] IR RC5(x) protocol handler initialized
[    1.167711] IR RC6 protocol handler initialized
[    1.172325] IR JVC protocol handler initialized
[    1.176868] IR Sony protocol handler initialized
[    1.181512] IR RC5 (streamzap) protocol handler initialized
[    1.187092] IR SANYO protocol handler initialized
[    1.191816] IR MCE Keyboard/mouse protocol handler initialized
[    1.198922] [UDOO power-off] Probed LVDS power.
[    1.203497] udoo_power_off poweroff.24: request of gpio lcd_panel_on_gpio 123 failed with -16
[    1.212183] [UDOO power-off] Probed LVDS backlight.
[    1.217092] udoo_power_off poweroff.24: request of gpio lcd_backlight_gpio 163 failed with -16
[    1.225798] [UDOO power-off] Probed sitronix st1232 touch panel.

root@udooneo:~# cat /dev/input/event1 | hexdump //(works in the same way with quad, just replace event1 by event0)
0000000 703b 56be b2ac 000c 0003 0030 0030 0000
0000010 703b 56be b2ac 000c 0003 0035 0080 0000
0000020 703b 56be b2ac 000c 0003 0036 0065 0000
0000030 703b 56be b2ac 000c 0000 0002 0000 0000
0000040 703b 56be b2ac 000c 0000 0000 0000 0000
0000050 703b 56be d45d 000c 0003 0030 0030 0000
0000060 703b 56be d45d 000c 0003 0035 0080 0000
0000070 703b 56be d45d 000c 0003 0036 0065 0000

root@udooneo:~# /usr/bin/ts_calibrate
xres = 800, yres = 480
tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)
Took 2 samples...
Top left : X = 997160367 Y = 1000388592
Took 2 samples...
Top right : X = 997160367 Y = 1000388592
Took 2 samples...
Bot right : X = 997160367 Y = 1000388592
Took 2 samples...
Bot left : X = 997160367 Y = 1000388592
Took 2 samples...
Center : X = 997160367 Y = 1000388592
-0.831649 -0.173352 0.172793
-0.722514 0.317711 -0.316686
Calibration constants: -54502 -11360 11324 -47350 20821 -20754 65536 
```

the i2c touch is recognize at boot and loaded with kernel. When i make a basic test with cat /dev/input/event1 | hexdump  i can see the event update when i touch the screen. But if i want to calibrate it or test it it didn't work! i can see the cursor but when i touch the screen it stay at the same position ! Qt advise to use tslib for embedded http://doc.qt.io/qt-5/embedded-linux.html

graugans, 2016-02-15:
Can you please test with 
```export TSLIB_TSDEVICE=/dev/input/event1
```
Acording to [ts_calibrate manpage](http://manpages.ubuntu.com/manpages/raring/man1/ts_calibrate.1.html)

modjo756, 2016-02-15:
Hi graugans, sorry but i already did this test (and retry it now), see under 

```
root@udooneo:~# export TSLIB_TSDEVICE="/dev/input/event1"
root@udooneo:~# export TSLIB_CONFFILE="/etc/ts.conf"
root@udooneo:~# /usr/bin/ts_calibrate
xres = 800, yres = 480
tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)
Took 5 samples...
Top left : X = 3160116 Y = 1979711542
Took 5 samples...
Top right : X = 3160116 Y = 1979711542
Took 5 samples...
Bot right : X = 3160116 Y = 1979711542
Took 5 samples...
Bot left : X = 3160116 Y = 1979711542
Took 5 samples...
Center : X = 3160116 Y = 1979711542
-0.037948 -0.849673 0.001356
0.027042 1.419819 -0.002266
Calibration constants: -2486 -55684 88 1772 93049 -148 65536 
root@udooneo:~# /usr/bin/ts_test
tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)
^Csignal 2 caught
root@udooneo:~# 

```
it seems that ts_calibrate works fine but not the test !

graugans, 2016-02-16:
I guess the error is 
```
tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)
```

According to this [stackoverflow](http://stackoverflow.com/questions/11937287/touchscreen-and-driver-installed-but-tslib-cannot-calibrate) you (we) have to patch the tslib library.

modjo756, 2016-02-16:
Yes this will be the solution, i think the second is better --> apply a patch during the build. Can you guide how to apply this patch, i will make a try !

graugans, 2016-02-17:
I do stuff like this mostly the [quilt way](http://www.yoctoproject.org/docs/2.0/dev-manual/dev-manual.html#using-a-quilt-workflow).

```
$ bitbake -cdevshell tslib
```
A new terminal window will be spawned pointing directly to the source of tslib.
```
# quilt new 0001-st1232-tslib.patch
# quilt add ./plugins/input-raw.c
# vim ./plugins/input-raw.c
# quilt refresh
# mkdir -p  /data/FSL/fsl-udoo-community-bsp/sources/meta-udoo/recipes-graphics/tslib/tslib/
# cp ../../../../../../../patches/0001-st1232-tslib
.patch /data/FSL/fsl-udoo-community-bsp/sources/meta-udoo/recipes-graphics/tslib/tslib/
```
After this a tslib_1.1.bbappend needs to be created

modjo756, 2016-02-17:
Great ! do you want i make this test or you commit the patch ?

graugans, 2016-02-17:
It would be great if you could test and if it works I think about how to add this in a Machine dependant and Yocto compatible way.

modjo756, 2016-02-17:
I think i didn't have the time before this week end (i'm busy with my job), can you wait ?

graugans, 2016-02-17:
Perfect. I am busy too so let's discuss this at the weekend. BTW the A62 boards are on the their way...

modjo756, 2016-02-17:
Cool, i'm so impatient to use your yocto layer with this card  ... I let you know before the end the week your patch ...

graugans, 2016-02-20:
Please check the last entry in my top post there is the fixed patch prefix

modjo756, 2016-02-21:
My file : [patch](https://gist.github.com/modjo756/036be569a27c2e5db4dd)

I'm sorry if i didn't understand ! 
now i have this error : 
ERROR: Command Error: exit status: 1  Output:
Applying patch 0001-st1232-tslib.patch
patching file plugins/input-raw.c
Hunk #2 FAILED at 99.
patch unexpectedly ends in middle of line
Hunk #3 FAILED at 204.
2 out of 3 hunks FAILED -- rejects in file plugins/input-raw.c
Patch 0001-st1232-tslib.patch does not apply (enforce with -f)
ERROR: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /media/modjo/data1TO/yocto/seco/udoo-community-bsp/neoBuild/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/tslib/1.1-r0/temp/log.do_patch.20608
ERROR: Task 919 (/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/meta/recipes-graphics/tslib/tslib_1.1.bb, do_patch) failed with exit code '1'

graugans, 2016-02-21:
Here we go
```patch
Index: fsl-udoo-community-bsp/build/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/tslib/1.1-r0/tslib-1.1/plugins/input-raw.c
===================================================================
--- tslib-1.1.orig/plugins/input-raw.c
+++ tslib-1.1/plugins/input-raw.c
@@ -43,6 +43,7 @@
 #ifndef ABS_MT_POSITION_X
 # define ABS_MT_POSITION_X       0x35    /* Center X ellipse position */
 # define ABS_MT_POSITION_Y       0x36    /* Center Y ellipse position */
+# define ABS_MT_POSITION_MAJOR	 0x30    /* Press */
 #endif
 
 #include "tslib-private.h"
@@ -98,8 +99,8 @@ static int check_fd(struct tslib_input *
 	if ((ioctl(ts->fd, EVIOCGBIT(EV_ABS, sizeof(absbit)), absbit)) < 0 ||
 		!(absbit[BIT_WORD(ABS_X)] & BIT_MASK(ABS_X)) ||
 		!(absbit[BIT_WORD(ABS_Y)] & BIT_MASK(ABS_Y))) {
-		fprintf(stderr, "tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)\n");
-		return -1;
+		//fprintf(stderr, "tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)\n");
+		//return -1;
 	}
 
 	/* Since some touchscreens (eg. infrared) physically can't measure pressure,
@@ -203,6 +204,7 @@ static int ts_input_read(struct tslib_mo
 				case ABS_MT_POSITION_Y:
 					i->current_y = ev.value;
 					break;
+				case ABS_MT_TOUCH_MAJOR:
 				case ABS_PRESSURE:
 					i->current_p = ev.value;
 					break;
```

modjo756, 2016-02-24:
Hi graugans, sorry but i have the same error with your file : 

ERROR: Command Error: exit status: 1  Output:
Applying patch 0001-st1232-tslib.patch
patching file plugins/input-raw.c
Hunk #2 FAILED at 99.
patch unexpectedly ends in middle of line
Hunk #3 FAILED at 204.
2 out of 3 hunks FAILED -- rejects in file plugins/input-raw.c
Patch 0001-st1232-tslib.patch does not apply (enforce with -f)
ERROR: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /media/modjo/data1TO/yocto/seco/udoo-community-bsp/neoBuild/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/tslib/1.1-r0/temp/log.do_patch.3720
ERROR: Task 919 (/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/meta/recipes-graphics/tslib/tslib_1.1.bb, do_patch) failed with exit code '1'

modjo756, 2016-02-25:
I find the mistake, in your last post at line @@ -98,8 +99,8 @@ static int check_fd(struct tslib_input * the ) is missing ! under the patch i used

```
Index: fsl-udoo-community-bsp/build/tmp/work/cortexa9hf-vfp-neon-poky-linux-gnueabi/tslib/1.1-r0/tslib-1.1/plugins/input-raw.c
===================================================================
--- tslib-1.1.orig/plugins/input-raw.c
+++ tslib-1.1/plugins/input-raw.c
@@ -43,6 +43,7 @@
 #ifndef ABS_MT_POSITION_X
 # define ABS_MT_POSITION_X       0x35    /* Center X ellipse position */
 # define ABS_MT_POSITION_Y       0x36    /* Center Y ellipse position */
+# define ABS_MT_POSITION_MAJOR  0x30    /* Press */
 #endif
 
 #include "tslib-private.h"
@@ -98,8 +99,8 @@
 	if ((ioctl(ts->fd, EVIOCGBIT(EV_ABS, sizeof(absbit)), absbit)) < 0 ||
 		!(absbit[BIT_WORD(ABS_X)] & BIT_MASK(ABS_X)) ||
 		!(absbit[BIT_WORD(ABS_Y)] & BIT_MASK(ABS_Y))) {
-		fprintf(stderr, "tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)\n");
-		return -1;
+		//fprintf(stderr, "tslib: Selected device is not a touchscreen (must support ABS_X and ABS_Y events)\n");
+		//return -1;
 	}
 
 	/* Since some touchscreens (eg. infrared) physically can't measure pressure,
@@ -203,6 +204,9 @@
 				case ABS_MT_POSITION_Y:
 					i->current_y = ev.value;
 					break;
+                                case ABS_MT_TOUCH_MAJOR:
+                                        i->current_p = ev.value;
+                                        break;
 				case ABS_PRESSURE:
 					i->current_p = ev.value;
 					break;
```

modjo756, 2016-02-25:
After a test, not work same problem with the touch ...

modjo756, 2016-02-26:
A test with evtest give this : 

```
root@udooneo:~# evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      20cc000.snvs-pwrkey
/dev/input/event1:      st1232-touchscreen
/dev/input/event2:      FreescaleAccelerometer
/dev/input/event3:      FreescaleMagnetometer
/dev/input/event4:      FreescaleGyroscope
Select the device event number [0-4]: 1
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "st1232-touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
  Event type 3 (EV_ABS)
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max      255
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      799
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      479
Properties:
Testing ... (interrupt to exit)
Event: time 1456491577.981717, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 50
Event: time 1456491577.981717, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 444
Event: time 1456491577.981717, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 200
Event: time 1456491577.981717, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1456491577.981717, -------------- SYN_REPORT ------------
Event: time 1456491577.990159, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 47
Event: time 1456491577.990159, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 444
Event: time 1456491577.990159, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 199
Event: time 1456491577.990159, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1456491577.990159, -------------- SYN_REPORT ------------
Event: time 1456491578.010773, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 44
Event: time 1456491578.010773, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 444
Event: time 1456491578.010773, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 196
Event: time 1456491578.010773, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1456491578.010773, -------------- SYN_REPORT ------------
Event: time 1456491578.018008, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 42
Event: time 1456491578.018008, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 444
Event: time 1456491578.018008, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 194
Event: time 1456491578.018008, ++++++++++++++ SYN_MT_REPORT ++++++++++++
```

graugans, 2016-02-27:
evtest seems to work with the touch display. Can you please provide the output off:

```bash
root@udooneo:~# export TSLIB_TSDEVICE="/dev/input/event1"
root@udooneo:~# export TSLIB_CONFFILE="/etc/ts.conf"
root@udooneo:~# /usr/bin/ts_calibrate
```

modjo756, 2016-02-28:
see under :  

`root@udooneo:~# export TSLIB_TSDEVICE="/dev/input/event1"
root@udooneo:~# export TSLIB_CONFFILE="/etc/ts.conf"
root@udooneo:~# /usr/bin/ts_calibrate
xres = 800, yres = 480
tslib: Selected device is not a touchscreen (must support BTN_TOUCH or BTN_LEFT events)
Took 5 samples...
Top left : X = 3160116 Y = 10567528
Took 5 samples...
Top right : X = 3160116 Y = 10567528
Took 5 samples...
Bot right : X = 3160116 Y = 10567528
Took 5 samples...
Bot left : X = 3160116 Y = 10567528
Took 5 samples...
Center : X = 3160116 Y = 10567528
-222.980225 1.544353 -0.461765
380.929413 -2.744118 0.820588
Calibration constants: -14613232 101210 -30262 24964590 -179838 53778 65536 
`

graugans, 2016-02-28:
Looks like more patching needed it complains about missing ``BTN_TOUCH`` events

graugans, 2016-02-28:
Maybe this patch from [timesys](http://repository.timesys.com/buildsources/t/tslib/tslib-1.1/tslib-1.1-fix-multitouch-support.patch) is the right direction?

modjo756, 2016-02-28:
I 'm building an image with this patch, i let you know ...

modjo756, 2016-02-28:
it works :+1: 

root@udooneo:/usr/share/quitbattery-1.0.0# /usr/bin/ts_calibrate
xres = 800, yres = 480
Took 7 samples...
Top left : X =   65 Y =  426
Took 4 samples...
Top right : X =  755 Y =  421
Took 8 samples...
Bot right : X =  743 Y =   50
Took 2 samples...
Bot left : X =   47 Y =   46
Took 8 samples...
Center : X =  408 Y =  223
1.693665 1.009995 -0.039999
476.071472 -0.000832 -1.010874
Calibration constants: 110996 66191 -2621 31199820 -54 -66248 65536

modjo756, 2016-02-28:
Now i have the touch on Qt app but, but it didn't calibrate (i make a ts calibrate before), need to fine more information about this ...

graugans, 2016-02-29:
So you've used only the timesys patch or did you have modified it somehow? So I can add this patch to the meta-udoo layer.

modjo756, 2016-02-29:
I use only the timesys patch.

modjo756, 2016-03-06:
After some tests this week end i can confirm that tslib is not needed to use lvds7 touch with neo ! I remove tslib from my local.conf. To use touch screen with Qt it needed to export : 
`QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event1:inverty`

inverty is because on qt y is reverse in some case (or need to rotate to 180 in other board !) :
[eglfs](http://doc.qt.io/qt-5/embedded-linux.html) see input section

Perhaps it's a good idea to add in your recipe in qt5-env.sh this 

`
export QT_QPA_EGLFS_PHYSICAL_HEIGHT=91
export QT_QPA_EGLFS_PHYSICAL_WIDTH=152
export QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS=/dev/input/event1:inverty
`

the dimensions is for 7inch touch

graugans, 2016-03-06:
Thanks for the update and the additional information at [udoo forum](http://www.udoo.org/forum/threads/openembedded-yocto-on-udoo-boards.2965/page-11#post-17548)

commit 3b48c86c (2016-03-13): Added MACHINE_FEATURE LVDS7

This enables Qt5.x.x touch support for the KIT LCD 7" Touch device.
To activate this feature add the following to your conf/local.conf

    MACHINE_FEATURES_append = " \
     lvds7 \
    "
This can not be activate as general option because not everybody owns a
7" touch screen panel from UDOO. A more sophisticated could be detecting
the screen based on the current device tree and setting those values
accordingly. This fixes #9
closed 2016-03-13 by graugans