graugans/meta-udoo #11: Master branch error with remove-libtool.bbclass

modjo756, 2016-03-16:
Trying to build an image in master Branch (and meta-qt5 -b jansa/master-5.6)

have this error with bitbake : 

modjo@Linux-corei5:/media/modjo/data1TO/yocto/seco/udoo-community-bsp/neoBuild$ MACHINE=udooneo bitbake udoo-image-qt5
Loading cache: 100% |#########################################################################################################################################################################################################| ETA:  00:00:00
Loaded 15 entries from dependency cache.
ERROR: ParseError at /media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/meta-qt5/classes/qmake5_base.bbclass:50: Could not inherit file classes/remove-libtool.bbclass

graugans, 2016-03-16:
Strange, I had this running on the weekend ``jansa/master-5.6``was not updated. Maybe ``poky`` or ``meta-openembedded`` are updated.

modjo756, 2016-03-16:
yes like me. I didn't use the same machine for this build , the first test i did it at my work this one at my home. i checkout all the meta to master, just meta-qt5 to jansa/master-5.6.

modjo756, 2016-03-16:
Ok, i removed my source folder and remake it with master branch, now i have this error (like you i suppose) : 

modjo@Linux-corei5:/media/modjo/data1TO/yocto/seco/udoo-community-bsp/neoBuild$ MACHINE=udooneo bitbake udoo-image-qt5
NOTE: Your conf/bblayers.conf has been automatically updated.
NOTE: Building Ne10 for armv7##################################################################                                                                                                                               | ETA:  00:00:33
Parsing recipes: 100% |#######################################################################################################################################################################################################| Time: 00:01:04
Parsing of 1981 .bb files complete (0 cached, 1981 parsed). 2540 targets, 415 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:
  /media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/meta-udoo/meta-python/recipes-devtools/python/python-m2crypto_0.21.1.bbappend

How did you resolve it ?

graugans, 2016-03-16:
I'll push the update

graugans, 2016-03-16:
Looks, like something is in limbo state on master :)

```
ERROR: gobject-introspection-1.46.0-r0 do_populate_sysroot: The recipe gobject-introspection is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /data/FSL/fsl-community-bsp-master/build/tmp/sysroots/udooneo/usr/share/gobject-introspection-1.0/Makefile.introspection
 Matched in manifest-udooneo-gtk-doc-stub.populate_sysroot
 /data/FSL/fsl-community-bsp-master/build/tmp/sysroots/udooneo/usr/share/aclocal/introspection.m4
 Matched in manifest-udooneo-gtk-doc-stub.populate_sysroot
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: gobject-introspection-1.46.0-r0 do_populate_sysroot: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
ERROR: gobject-introspection-1.46.0-r0 do_populate_sysroot: Function failed: sstate_task_postfunc
ERROR: Logfile of failure stored in: /data/FSL/fsl-community-bsp-master/build/tmp/work/cortexa9hf-neon-poky-linux-gnueabi/gobject-introspection/1.46.0-r0/temp/log.do_populate_sysroot.16494
ERROR: Task 1802 (/data/FSL/fsl-community-bsp-master/sources/poky/meta/recipes-gnome/gobject-introspection/gobject-introspection_1.46.0.bb, do_populate_sysroot) failed with exit code '1'
``` 

I asked for some help on this on the mailinglist

modjo756, 2016-03-16:
i didn't have the same error, i have this now (after your update) : 

```
modjo@Linux-corei5:/media/modjo/data1TO/yocto/seco/udoo-community-bsp/neoBuild$ MACHINE=udooneo bitbake udoo-image-qt5 
ERROR: Traceback (most recent call last):
  File "/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/bitbake/lib/bb/cookerdata.py", line 178, in wrapped
    return func(fn, *args)
  File "/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/bitbake/lib/bb/cookerdata.py", line 200, in parse_config_file
    return bb.parse.handle(fn, data, include)
  File "/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/bitbake/lib/bb/parse/__init__.py", line 113, in handle
    return h['handle'](fn, data, include)
  File "/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 124, in handle
    abs_fn = resolve_file(fn, data)
  File "/media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/bitbake/lib/bb/parse/__init__.py", line 136, in resolve_file
    raise IOError(errno.ENOENT, "file %s not found" % fn)
IOError: [Errno 2] file /media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/poky/meta-poky/conf/layer.conf not found
```

My bblayers.conf
```
POKY_BBLAYERS_CONF_VERSION = "1"

BBPATH = "${TOPDIR}"
BSPDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../..')}"

BBFILES ?= ""
BBLAYERS = " \
${BSPDIR}/sources/poky/meta \
${BSPDIR}/sources/poky/meta-poky \
\
${BSPDIR}/sources/meta-openembedded/meta-oe \
${BSPDIR}/sources/meta-openembedded/meta-multimedia \
${BSPDIR}/sources/meta-openembedded/meta-python \
${BSPDIR}/sources/meta-openembedded/meta-networking \
${BSPDIR}/sources/meta-openembedded/meta-ruby \
\
${BSPDIR}/sources/meta-fsl-arm \
${BSPDIR}/sources/meta-fsl-arm-extra \
${BSPDIR}/sources/meta-udoo \
${BSPDIR}/sources/meta-fsl-demos \
${BSPDIR}/sources/meta-qt5 \
"
```

graugans, 2016-03-17:
I am short in time, the ``meta-udoo`` should be the last layer in your bblayers file.

modjo756, 2016-03-17:
Sorry but i'm trying to remake all from scratch (git sources, checkout all branches to master except meta-qt5 ...) re-create with build folder and have this error : 

Your build environment has been configured with:

    MACHINE=udooneo
    SDKMACHINE=i686
    DISTRO=poky
    EULA=
modjo@Linux-corei5:/media/modjo/data1TO/yocto/seco/udoo-community-bsp/neoBuild$ MACHINE=udooneo bitbake udoo-image-qt5 -k
ERROR: ParseError at /media/modjo/data1TO/yocto/seco/udoo-community-bsp/sources/meta-qt5/classes/qmake5_base.bbclass:50: Could not inherit file classes/remove-libtool.bbclass                                                | ETA:  --:--:--

my local.conf 

`MACHINE ??= 'udooneo'
DISTRO ?= 'poky'
PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
PACKAGECONFIG_append_pn-qtbase = " accessibility pulseaudio"
PACKAGECONFIG_append_pn-qtmultimedia = " gstreamer"
ASSUME_PROVIDED += "libsdl-native"
CONF_VERSION = "1"

DL_DIR ?= "${BSPDIR}/downloads/"
ACCEPT_FSL_EULA = "1"

QT5_FULL = "\
    gstreamer \
    gst-plugins-base-app \
    gst-plugins-base \
    gst-plugins-good \
    gst-plugins-good-rtsp \
    gst-plugins-good-udp \
    gst-plugins-good-rtpmanager \
    gst-plugins-good-rtp \
    gst-plugins-good-video4linux2 \
    qtcanvas3d \
    qtwebengine \
    qtwebengine-qmlplugins \
    qtgraphicaleffects \
    qtlocation \
    qtquick1 \
    qtquick1-qmlplugins \
    qtquick1-plugins \
    qtserialport \
    qtquickcontrols-qmlplugins \
    qtserialbus \
    qttools-plugins \
    qtxmlpatterns \
    gstreamer \
    gst-plugins-base-app \
    gst-plugins-base \
    gst-plugins-good \
    gst-plugins-good-rtsp \
    gst-plugins-good-udp \
    gst-plugins-good-rtpmanager \
    gst-plugins-good-rtp \
    gst-plugins-good-video4linux2 \
    quitbattery \
    qt5ledscreen \
    evtest\
    "

IMAGE_INSTALL_append = "\
    htop \
    nano \
    cpufrequtils \
    ${QT5_FULL} \
    "
CORE_IMAGE_EXTRA_INSTALL = "packagegroup-core-sdk"

BUILD_LDFLAGS_prepend_pn-qemu-native = "-Wl,-rpath-link,/lib/x86_64-linux-gnu "

DISTRO_FEATURES_remove = "x11 wayland"

MACHINE_FEATURES_append = " \
 lvds7 \
"`

my bblayers
`LCONF_VERSION = "6"

BBPATH = "${TOPDIR}"
BSPDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../..')}"

BBFILES ?= ""
BBLAYERS = " \
  ${BSPDIR}/sources/poky/meta \
  ${BSPDIR}/sources/poky/meta-yocto \
  \
  ${BSPDIR}/sources/meta-openembedded/meta-oe \
  ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
  ${BSPDIR}/sources/meta-openembedded/meta-python \
  ${BSPDIR}/sources/meta-openembedded/meta-networking \
  \
  ${BSPDIR}/sources/meta-fsl-arm \
  ${BSPDIR}/sources/meta-fsl-arm-extra \
  ${BSPDIR}/sources/meta-fsl-demos \
  \
  ${BSPDIR}/sources/meta-qt5 \
  ${BSPDIR}/sources/meta-openembedded/meta-ruby \
  \
  ${BSPDIR}/sources/meta-udoo \
"`

Sorry but i'm stuck for the moment ...

modjo756, 2016-03-17:
try to use master branch for meta qt5, build start : 

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-15.10"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "udooneo"
DISTRO            = "poky"
DISTRO_VERSION    = "2.0.1"
TUNE_FEATURES     = "arm armv7a vfp thumb neon callconvention-hard cortexa9"
TARGET_FPU        = "vfp-neon"
meta              
meta-yocto        = "master:1a52eceaa5df89914b6a711defdcf0046e74c7f6"
meta-oe           
meta-multimedia   
meta-python       
meta-networking   = "master:dc5634968b270dde250690609f0015f881db81f2"
meta-fsl-arm      = "master:8d22b44716fa624ff87383c36d222d9e28d0b267"
meta-fsl-arm-extra = "master:2fdd1f7729c11900126f6c73346b8ea89448e927"
meta-fsl-demos    = "master:8bffde8d803dd2362fbded79781ce084d723b048"
meta-qt5          = "master:d6fe20e6083d421cc6091892e1d302c46e2d075d"
meta-ruby         = "master:dc5634968b270dde250690609f0015f881db81f2"
meta-udoo         = "master:0ca56fc886b65776c2811b56de242bb2a628df01"

graugans, 2016-05-13:
Due to the fact meta-qt5 switched to qt5.7 recently I have to update stuff taking this into account

graugans, 2016-10-19:
I'll close this due to the fact master is not supported at the moment.

closed 2016-10-19 by graugans