graugans/meta-udoo #12: Problem with can bus

modjo756, 2016-03-19:
When we use a dtb (produce with udoobuntu and device tree editor) file that enable flexcan on neo pin, they have some error during the boot : 

see : [flexcan error](https://gist.github.com/modjo756/bff1cd6522d1beef92b5)

and : [boot log](https://gist.github.com/modjo756/2760de1b66d53fef0989)

when we use ifconfig -a, didn't see can0 or can1 like [udoobuntu](http://www.udoo.org/forum/threads/canopen-communication-with-qt-on-udoo-neo.3709/#post-17857)

graugans, 2016-03-20:
Sorry,  I am busy with **real world** tasks at the moment I will try to take a look into this soon.

modjo756, 2016-03-20:
No problem, yes it takes times to play with this cards ...
This is the dts files from udoobuntu (in opt/dtweb/dtbkernel/arch/arm/boot/)
[dts-udoobuntu_CANok_Uart6.tar.gz](https://github.com/graugans/meta-udoo/files/181235/dts-udoobuntu_CANok_Uart6.tar.gz)

modjo756, 2016-03-20:
Ok, i have copied the folder dtbkernel from udoobuntu to my yocto sd card (opt folder). boot the card with yocto and go to dts. I "make dtbs" and copy the dtb generated to /boot/dts-overlay. reboot the card and same problem, no can appears and same problem during the boot ...

graugans, 2016-03-20:
To use CAN you'll have to disable the ``M4`` core by adding this to your ``uEnv.txt``

```
# Enable or Disable the Cortex M4 CPU
# m4_enabled=true|false
# any other value (or if the variable is not specified) will default to "true"
m4_enabled=false
```

After this I get the CAN interface and there is no more kernel OOPS. Can you please check if in UDOOBuntu the m4 is as-well disabled?

modjo756, 2016-03-20:
I will try this but on udoobuntu there is M4 enabled (arduino working) and CAN enabled ...

graugans, 2016-03-20:
So could you please so kind and call ``md5sum`` on the m4 firmware files. Maybe I use old ones

modjo756, 2016-03-20:
I suppose you talk about the files on boot folder ? 

In udoobuntu : 

_modjo@Linux-corei5:/media/modjo/boot1$ md5sum m4clean.fw 
7aa83b6b25b23cc7d6e92aa757a0920e  m4clean.fw
modjo@Linux-corei5:/media/modjo/boot1$ md5sum m4startup.fw 
32a0519c167f98070956e601764c71df  m4startup.fw_

In meta-udoo master

_modjo@Linux-corei5:/media/modjo/boot2$ md5sum m4clean.fw 
7aa83b6b25b23cc7d6e92aa757a0920e  m4clean.fw
modjo@Linux-corei5:/media/modjo/boot2$ md5sum m4startup.fw 
32a0519c167f98070956e601764c71df  m4startup.fw
modjo@Linux-corei5:/media/modjo/boot2$ _

graugans, 2016-03-20:
I am confused, the CAN interface is only initialized properly when M4 is disabled. This needs more investigation.

modjo756, 2016-03-20:
Yes this strange ... otherwise qtserialbus[ working on neo ](http://www.udoo.org/forum/threads/canopen-communication-with-qt-on-udoo-neo.3709/#post-17904)

modjo756, 2016-04-01:
Hi graugan, for the problem with can and M4, perhaps we can use this tool tool to enable the M4 after  [start](http://www.udoo.org/docs-neo/Cookbook_Linux/Turn_off_M4_Arduino_core.html) ? Not present for the moment in yocto. What do you think about this ?

graugans, 2016-04-01:
You have an idea where the source is located for this? I guess this is some php5 script.

modjo756, 2016-04-01:
No i need to investigate ...

graugans, 2016-04-01:
here we go [https://github.com/fmntf/dtweb/blob/master/utils/udoom4ctl](https://github.com/fmntf/dtweb/blob/master/utils/udoom4ctl) it is a PHP5  app. It just sets the value in the ´´uEnv.txt´´

modjo756, 2016-04-02:
Thanks, this not resolve the problem ...

graugans, 2016-04-03:
I have a hunch, it smells like we have to update the kernel version. I just tested with a vanilla UDOOBuntu -> same error like on yocto....

graugans, 2016-04-03:
No chance on UDOO Neo with UDOObuntu and latest kernel same issues like on Yocto. When M4 is disabled CAN works like a charm :(

graugans, 2016-04-03:
I guess I've fixed the CAN issue:

```
root@udooneo:~# ifconfig -a
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:142 

can1      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:143 

eth0      Link encap:Ethernet  HWaddr 00:c0:08:8c:2d:86  
          inet addr:192.168.23.125  Bcast:192.168.23.255  Mask:255.255.255.0
          inet6 addr: fe80::2c0:8ff:fe8c:2d86/64 Scope:Link
          inet6 addr: fd6f:be9c:6c36:0:2c0:8ff:fe8c:2d86/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:23 errors:0 dropped:1 overruns:0 frame:0
          TX packets:56 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3090 (3.0 KiB)  TX bytes:10106 (9.8 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:140 (140.0 B)  TX bytes:140 (140.0 B)

wlan0     Link encap:Ethernet  HWaddr de:ad:be:ef:00:00  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@udooneo:~# 
```

ATM it looks like a combination of the default M4 firmware, kernel and DeviceTree. After using this sketch as initial FW everything is okay:

```
void setup(){
    Serial.begin(115200);
    Serial0.begin(115200);
    
    pinMode(13, OUTPUT);
}

void loop(){
    Serial.print("Hello");
    Serial.print(" ");
    Serial.println("A9!");
   
    digitalWrite(13, HIGH);
    delay(1000);
    
    Serial0.print("Hello");
    Serial0.print(" ");
    Serial0.println("world!");
    
    digitalWrite(13, LOW);
    delay(1000);
}
```

But now its time for the couch...

modjo756, 2016-04-04:
Congratulation Graugans, i did the same ! I added a sketch on /var/opt/m4 (blink example) and now it works CAN+M4. Need to add a default sketch to yocto ...

graugans, 2016-04-05:
I've uploaded one yesterday:https://github.com/graugans/meta-udoo/commit/b873cbfef36f64aa3337a4c938329b3a21cab101

modjo756, 2016-05-03:
Problem solved with your add

closed 2016-05-03 by modjo756