graugans/meta-udoo #13: Add bluez support for qt5

modjo756, 2016-05-03:
Bluetooth work on yocto but to use it with qt, we need to add bluez 5 before the build of qt. Looking to add this bluez5 on neo.

graugans, 2016-05-03:
This is something waiting on the TODO list :) I wrote this [howto on bluez5](http://ch.ege.io/blog/2015/04/12/running-bluez5-on-yocto-slash-openembedded/) a year ago I am not sure if this is still valid.

modjo756, 2016-05-03:
Hi graugans, yes i tried to add this : 

MACHINE_FEATURES_append = " \
 lvds7 \
 bluetooth \
 bluez5 \
"
IMAGE_INSTALL_append = "\
    htop \
    nano \
    cpufrequtils \
    ${QT5_FULL} \
    bluez5 \
    bluez5-testtools \
    bluez5-noinst-tools \
    "

it build : 
Currently 1 running tasks (1053 of 1060):
0: bluez5-5.33-r0 do_compile (pid 9434)

i also find this recipe : [http://layers.openembedded.org/layerindex/recipe/39156/](url) but i don't know if it realy necessary ...

when i launch a qt exemple (scanner bluetooth) i have this warning : 
"Cannot find a running Bluez. Please check the Bluez installation."

Perhaps i need to rebuild the image from scratch (clean qt build ...) !

modjo756, 2016-05-04:
It seems to work (i can discover device with Qt), but i need to launch manually  bluetoothd  in /usr/lib/bluez5. Do you have an idea how to configure yocto to launch it at startup ? I need to make more test (bluetoothLE ...) to check if all work fine.

graugans, 2016-05-04:
I have to check about bluetoothd maybe systemd image is expected.. related to BLE I did some [BLE test last year](http://ch.ege.io/blog/2015/04/12/orico-bta-402-usb-bluetooth-4-dot-0-low-energy-micro-adapter-on-odroid-c1/) It was on the ODROID-C1 but this shouldn't have changed much

modjo756, 2016-05-06:
Good news, bluetooth LE seems to work with Qt --> test with heart rate example and my geonaute heart belt, works fine ! Now need to launch bluetoothd at startup :-)

graugans, 2016-05-08:
Sorry for the delay, but the weather is far too beautiful to sit in the basement :)

The bluetooth daemon is disabled by default, when reading the init script things get clear

```bash
# FIXME: any of the sourced files may fail if/with syntax errors
test -f /etc/default/bluetooth && . /etc/default/bluetooth
test -f /etc/default/rcS && . /etc/default/rcS

set -e

case $1 in
  start)
        echo "Starting $DESC"

        if test "$BLUETOOTH_ENABLED" = 0; then
                echo "disabled. see /etc/default/bluetooth"
                exit 0
        fi

        start-stop-daemon --start --background $SSD_OPTIONS
        echo "${DAEMON##*/}"

  ;;
```

When you provide a file ``/etc/default/bluetooth`` with the following content:

```bash
BLUETOOTH_ENABLED=1
``` 

``bluetoothd`` should fire up during start-up

modjo756, 2016-05-09:
Thanks, i have create the file /etc/default/bluetooth with BLUETOOTH_ENABLED=1 but nothing change at startup. Do i need to create the script you talk above in rc.local or another place ?

graugans, 2016-05-09:
Looks like there is another bug in the init file the entry

```
DAEMON=/usr/lib/bluez5/bluetooth/bluetoothd
```
is invalid. The right one is:
```
DAEMON=/usr/libexec/bluetooth/bluetoothd
```

modjo756, 2016-05-10:
Thanks, but can you tell me where is locate the init file please ?

graugans, 2016-05-10:
Sure, ``/etc/init.d/bluetoothd``

modjo756, 2016-05-10:
Thanks, i didn't have this file in my sdcard, i suppose i need to create it ? after that do i need to export (or register ...) it to enable the service at startup ? Otherwise, i just saw that bluez is already install in master image (not tested jethro), it's not necessary to add  bluez to local.conf ...

graugans, 2016-05-12:
I did exactly what I described [here](http://ch.ege.io/blog/2015/04/12/orico-bta-402-usb-bluetooth-4-dot-0-low-energy-micro-adapter-on-odroid-c1/) this provides the bluez5 stuff. [I've submitted a patch](http://patches.openembedded.org/patch/122321/) to oe-core for review which fixes the init script issue.

graugans, 2016-05-20:
My [patch](https://github.com/openembedded/openembedded-core/commit/63cde9b8101ed39376529f3205d32729341d5802) is applied upstream. So everything should work as intended.

modjo756, 2016-05-25:
Ok, i launch a new build with krogoth branch, but to apply your patch i merge poky to master branch.

graugans, 2016-05-25:
I can add the fix for meta-udoo and krogoth. Maybe I can try in parallel to bring it in krogoth branch of openembedded aswell. As a next patch I think we should push the ``qtserialbus`` patch to meta-qt5. I am still in the progress to fine tune the krogoth branch of meta-udoo.

modjo756, 2016-05-25:
bitbake bluez5 with udoo krogoth and master poky works fine. Now i have launch udoo-image-qt5 build. I let you know. Yes it would be a good thing to add a patch for qtserialbus to meta-qt5 !

modjo756, 2016-05-25:
Build with krogoth ok, just somes errors with md5 checksum see [here](https://gist.github.com/modjo756/f808bf4e826fc041a9be0ae8fb7bf771)

modjo756, 2016-05-26:
All work fine with krogoth. Bluetooth is started at boot and work with Qt. Arduino, CAN, uart6 (with patch) works all ! Good work :+1:

closed 2016-05-26 by modjo756