graugans/meta-udoo #14: Rx on uart6 not working ...

modjo756, 2016-05-12:
like describe in this [thread](http://www.udoo.org/forum/threads/a9-uart6-and-udoobuntu.3196/), uart6 not work, Rx FIFO overrun. To correct this we need to rebuild the kernel with this patch 
[udooUart6.patch.zip](https://github.com/graugans/meta-udoo/files/261558/udooUart6.patch.zip) and add file://udooUart6.patch \ to linux-udooboard.inc. Now work fine :-)

graugans, 2016-05-13:
I'll fix this by end of next week when getting back from vacation

graugans, 2016-05-13:
You still see this issue? Because the [PR on UDOOBoard kernel](https://github.com/UDOOboard/linux_kernel/pull/16) is closed.

modjo756, 2016-05-17:
Thanks for your link, i let a message about this issue on this thread !

modjo756, 2017-04-21:
Uart6 works fine now.

closed 2017-04-21 by modjo756