graugans/meta-udoo #15: Krogoth branch and Qt5.7

modjo756, 2016-07-09:
Hi graugans, i have compiled yocto krogoth branch with meta-qt5 in master branch to have qtchart ( a new module to draw charts ...) It's compile fine but i need to remove  qtbase-fonts  in your udoo-image-qt5.bb to remove an error --> apparently in qt5.7 they remove it ! but now i have an error when i deploy an app on the card (in neo for the moment) : 

_QFontDatabase: Cannot find font directory /usr/lib/fonts.
Note that Qt no longer ships fonts. Deploy some (from http://dejavu-fonts.org for example) or switch to fontconfig.
EGL Error : Could not create the egl surface: error = 0x300b_

Do you know how i can resolve this ?

i found that on the web : 
[qtFont](http://lists.openembedded.org/pipermail/openembedded-devel/2016-April/106780.html)

graugans, 2016-07-10:
I will take a look when I've finished Seco A62 GLES and A62 CAN & Neo CAN

modjo756, 2016-07-10:
Ok no problem ! I will post a message on mailing list to see if someone have a solution !

modjo756, 2016-10-12:
I have make some progress today on this point. On my host machine i have downloaded [dejavu](http://dejavu.sourceforge.net). On the card i have created a folder fonts in /usr/lib/. with scp, copied the dejavu fonts in this folder, now qt works fine !
To integrate this layer on qt, i added ttf-dejavu in local.conf like this : 

IMAGE_INSTALL_append = "\
     htop \
     nano \
     ttf-dejavu \
"
bitbake and build start ... error during the rootfs : 
ERROR: ttf-dejavu not found in the base feeds

Do you have an idea what is the problem ? How we can install ttf-dejavu in /usr/lib/fonts ?

modjo756, 2016-10-13:
update with the help of yocto mailling list, i had this in my local.conf 
    ttf-dejavu-sans \
    ttf-dejavu-sans-mono \
    ttf-dejavu-sans-condensed \
    ttf-dejavu-serif \
    ttf-dejavu-serif-condensed \
    ttf-dejavu-common \

now it build without any error ! Now the problem is to install the ttf font in the folder /usr/lib/fonts ... and i didn't know how to do this !

modjo756, 2016-10-13:
i have tried this : 
add a folder recipe-graphics/ttf-dejavu in meta udoo.
add this file in ttf-dejavu
[ttf-dejavu_2.35.bbappend.tar.gz](https://github.com/graugans/meta-udoo/files/527604/ttf-dejavu_2.35.bbappend.tar.gz)
 but it seems not install the files in the good place ...

graugans, 2016-10-14:
Maybe [this](http://www.qtforum.org/article/25367/embedding-custom-fonts.html) is another option?

modjo756, 2016-10-14:
I will take a look to this today or this week end.

modjo756, 2016-10-15:
i have resolve the problem. See my folder [here](https://github.com/modjo756/meta-udoo-modjo/tree/krogoth-modjo/recipes-graphics) . Let me know if you would a pull request

modjo756, 2016-10-19:
I close this issue because it work for me now on quad and sa62 with qt5.7

closed 2016-10-19 by modjo756