graugans/meta-udoo #16: Arduino not working on udoo quad

modjo756, 2016-08-22:
For the moment it's impossible to program the sam3x of udoo quad with yocto, i have this error : 
SAM-BA operation failed
ERASE FLASH
need help to correct this !

graugans, 2016-08-26:
Sorry UDOO board is out of reach for the next weeks could you ping me in three weeks?

modjo756, 2016-08-30:
Ok no problem !

graugans, 2016-10-15:
After adding the [bossac](https://github.com/graugans/meta-udoo/commit/c696609b3ca01aaf9dc94ebdaf8a4b42c3a858a2) I can manually erase the ``SAM3X``

```
root@udooqdl:~# bossac-udoo -e --port=ttymxc3                                                                                                                                                               
Erase flash
```

graugans, 2016-10-15:
We might need to check the status of the following PINS in the device-tree

```
ERASE - P24 - DISP0_DAT0
IMX6_PIC_RST - T5 GPIO_0

graugans, 2016-10-19:
I'll close this issue. If the issue is still present feel free to reopen it again. For me it worked with the [following workarround](http://www.udoo.org/docs/Arduino_DUE_%28Sam3x%29/Troubleshooting_Programming_Arduino_Due_from_External_PC.html)

closed 2016-10-19 by graugans

modjo756, 2016-10-19:
@graugans just testted, work fine --> good job :+1: