graugans/meta-udoo #17: SECOA62 lvds7 screen not working

modjo756, 2016-09-19:
Hi, i have rebuilt a complete image (from scratch) for secoA62 Krogoth branch. All work fine but lvds7 is not working, see this post for details : http://www.udoo.org/forum/threads/openembedded-yocto-on-udoo-boards.2965/page-18#post-21027

modjo756, 2016-10-07:
I have make a patch to use the lvds7 on secoA62, see here : 
[patchLvds7](https://gist.github.com/modjo756/183251fa29399b42f3397d41ab16a1b1)

To have a working screen, i removed the second channel of the lvds (i don't know the reason of this bug ...) 

In this patch, i already activate some devices : 
- spi2
- can1
- i2c1
- uart4

All seems to be working fine (i tested all of this). The point that need to be improved is audio. If i uncomment ssi1 on the patch (line 337) i have some errors during the build of the kernel ...
Another point is the power button, not working ...
Graugans, i don't know if you plan to use a new version of kernel (4.1 ?) but it could be a base !

graugans, 2016-10-07:
Audio is on my todo  as-well I am unsure in what shape 4.1 for A62 is. A first step should be to bring device tree of A62 to the same level of udoo quad/dual. Then porting is more easy

graugans, 2016-10-14:
@modjo756 could you please be so kind and add a Pull request for the ``CONFIG_SPI_SPIDEV=y`` in defconfig? 

Kernel 4.1 is definitely [on the list](https://github.com/graugans/meta-udoo/issues/18). I would like to play with device-tree overlays. I am not sure of there is support in the 4.1 NXP kernel or not.

modjo756, 2016-10-14:
Yes i will do it this week end. Do i add my [patch](https://gist.github.com/modjo756/183251fa29399b42f3397d41ab16a1b1)  in the same time ?

graugans, 2016-10-14:
For the patch I would like to go like I did in UDOO Neo. Or rework the whole selection of configuration. Like device-tree overloads

modjo756, 2016-10-14:
ok no problem i understand. I let my patch in my git.

modjo756, 2016-11-15:
issue close with the update of u-boot-fslc. Now working fine

closed 2016-11-15 by modjo756