graugans/meta-udoo #18: Add a kernel recipe for kernel 4.1

graugans, 2016-09-22:
Thanks to @fmntf there is a kernel 4.1 port available for [udoo boards](https://github.com/UDOOboard/linux_kernel/issues/23) the branch is located in his [kernel repo](https://github.com/fmntf/linux_kernel/tree/4.1.15_1.x-udoo)

modjo756, 2016-12-15:
Hi garugans, just to know if you have made some progress on this ? I saw on[ udoo github ](https://github.com/UDOOboard/linux_kernel/issues/23#issuecomment-267331976)that they make some progress

graugans, 2016-12-15:
Not startend so no progress ;) for quad there are even mainline patches available. But in this case etnaviv will be needed for opengl. I do not know if this works with QT5

cyrilfr, 2017-01-31:
I moved forward to morty and I added the u-boot-fscl and the linux-imx (kernel v4.1.15) recipes from meta-freescale to my project. It does compile for the UDOO NEO (I dropped the Quad) but I stay blocked at boot:

U-Boot SPL 2016.11+fslc+gc44711d (Jan 31 2017 - 09:26:33)
Setting 1024MB RAM calibration data
Trying to boot from MMC1
port 1

U-Boot 2016.11+fslc+gc44711d (Jan 31 2017 - 09:26:33 +0100)

CPU:   Freescale i.MX6SX rev1.2 at 792 MHz
Reset cause: POR
Board: UDOO Neo Full
I2C:   ready
DRAM:  1 GiB
PMIC:  PFUZE3000 DEV_ID=0x30 REV_ID=0x11
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   using phy at 0
FEC0 [PRIME]
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
701 bytes read in 411 ms (1000 Bytes/s)
Running bootscript from mmc ...
M4: disabled via environment variables.
Device Tree: /boot/imx6sx-udoo-neo-full-lvds7.dtb
4325088 bytes read in 906 ms (4.6 MiB/s)
Booting from mmc ...
46795 bytes read in 624 ms (72.3 KiB/s)
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
   Using Device Tree in place at 83000000, end 8300e6ca

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.15-2.0.0+gb63f3f5 (eeproperty@eeproperty-OptiPlex-9020) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT Tue Jan 31 10:57:24 CET 2017
[    0.000000] CPU: ARMv7 Processor [412fc09a] revision 10 (ARMv7), cr=50c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: UDOO Neo Full
[    0.000000] Reserved memory: created CMA memory pool at 0xac000000, size 320 MiB
[    0.000000] Reserved memory: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] PERCPU: Embedded 11 pages/cpu @ab728000 s16332 r8192 d20532 u45056
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260096
[    0.000000] Kernel command line: console=ttymxc0,115200 root=/dev/mmcblk0p1 rootwait rw rootfstype=ext4 uart_from_osc clk_ignore_unused cpuidle.off=1 consoleblank=0
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 704024K/1048576K available (4063K kernel code, 260K rwdata, 2044K rodata, 204K init, 389K bss, 16872K reserved, 327680K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xc0800000 - 0xff000000   (1000 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xc0000000   (1024 MB)
[    0.000000]     pkmap   : 0x7fe00000 - 0x80000000   (   2 MB)
[    0.000000]     modules : 0x7f800000 - 0x7fe00000   (   6 MB)
[    0.000000]       .text : 0x80008000 - 0x805fef20   (6108 kB)
[    0.000000]       .init : 0x805ff000 - 0x80632000   ( 204 kB)
[    0.000000]       .data : 0x80632000 - 0x806732a0   ( 261 kB)
[    0.000000]        .bss : 0x80676000 - 0x806d76dc   ( 390 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=1.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[... 125 lines cut ...]
[    1.944474] caam algorithms registered in /proc/crypto
[    1.953768] caam_jr 2101000.jr0: registering rng-caam
[    1.959186] snvs-secvio 20cc000.caam-snvs: can't identify tamper alarm configuration
[    1.966954] snvs-secvio: probe of 20cc000.caam-snvs failed with error -22
[    1.977433] fsl-asrc 2034000.asrc: driver registered
[    1.985422] nf_conntrack version 0.5.0 (16120 buckets, 64480 max)
[    1.992289] NET: Registered protocol family 17
[    1.996801] Key type dns_resolver registered
[    2.002020] cpu cpu0: dev_pm_opp_get_opp_count: device OPP not found (-19)
[    2.028250] failed to register cpuidle driver
[    2.032627] ThumbEE CPU extension supported.
[    2.039750] snvs_rtc 20cc000.snvs:snvs-rtc-lp: setting system clock to 1970-01-01 00:00:00 UTC (0)
[    2.059864] VLDO2: disabling
[    2.063224] wlan-en-regulator: disabling
[    2.067155] vref-3v3: disabling
[    2.070333] PSU-5V0: disabling
[    2.073393] usb_otg2_vbus: disabling
[    2.076971] usb_otg1_vbus: disabling
[    2.080787] vddpcie-phy: disabling
[    2.084250] clk: Not disabling unused clocks
[    2.088747] ALSA device list:
[    2.091726]   No soundcards found.
[    2.095374] Waiting for root device /dev/mmcblk0p1...

Any idea?

fmntf, 2017-01-31:
That issue if fixed on my kernel: https://github.com/fmntf/linux_kernel/commit/0dffbfaa5b5877ec32bd31c709eefe1e8c713358

cyrilfr, 2017-01-31:
Thanks @fmntf but I already use your DTS and defonconfig files. I guess this is an issue either with u-boot-fslc or the wic image.

graugans, 2017-01-31:
@cyrilfr This is not a problem with the wic tool. It looks like an issue with the mapping of the mmc. From your kernel log the mmc seems to be detected:

```
[    1.408251] mmc2: SDHCI controller on 2198000.usdhc [2198000.usdhc] using ADMA
[    1.422345] mmc0: error -110 whilst initialising SD card
```
The Mapping is done [imx6sx.dtsi#L35](https://github.com/cyrilfr/meta-udoo/blob/morty/recipes-kernel/linux/linux-imx/udooneo/dts/imx6sx.dtsi#L35)

U-Boot sets the root path to 
```
mmcroot=/dev/mmcblk0p1 rootwait rw
```

cyrilfr, 2017-01-31:
@graugans Yes you're right and as you can see U-Boot loads the right DTB file so it should map the mmc0. I've never been able to compile the sources from the fork https://github.com/fmntf/linux_kernel/tree/4.1.15_1.x-udoo . I wonder if this one is able to boot.

fmntf, 2017-01-31:
Tomorrow I will try to rebuild it. I also have a newever kernel (4.1.15_2.0.0) on another computer, but I have forgotten to push it to GitHub.
Probably newer compilers broke something. On Ubuntu 16.04 with the default cross-compiler it worked fine.

I used U-boot 2015 and 2016. They both worked fine.

cyrilfr, 2017-01-31:
I tried with u-boot-fslc from [this recipe](https://github.com/Freescale/meta-freescale/blob/master/recipes-bsp/u-boot/u-boot-fslc_2016.11.bb) of meta-freescale. I also compiled the kernel with [linux-imx](https://github.com/Freescale/meta-freescale/blob/master/recipes-kernel/linux/linux-imx_4.1.15.bb) from meta-freescale layer. There is u-boot-imx and u-boot-fslc, linux-imx and linux-fslc recipes actually. I'll have a look on what are the differences. This is probably more clever to use u-boot-imx with linux-imx or your fork if it comes from linux-imx.

graugans, 2017-02-01:
@cyrilfr It should not really depend on the U-Boot. The kernel is orchestrated by the device-tree, the only influence U-Boot does have in this case is the ``bootargs`` the difference between  ``linux-imx`` and linux-fslc is the ``c`` what stands for community. The [linux-fslc](https://github.com/Freescale/linux-fslc) is maintained by the community and hosted at [github](https://github.com/Freescale/linux-fslc). The linux-imx is the official kernel source hosted at Freescale/NXP. I would go with the linux-fslc sources. Because there maybe the chance we can integrate the Neo stuff.

cyrilfr, 2017-02-01:
Yes so I edited my recipe to use linux-fslc instead. The kernel compiles but I have an error while compiling the kernel-module-imx-gpu-viv_5.0.11 module.

```
kernel-module-imx-gpu-viv/5.0.11.p8.6+fslc+gitAUTOINC+1be63fb14e-r0/git/kernel-module-imx-gpu-viv-src/hal/os/linux/kernel/platform/freescale/gc_hal_kernel_platform_imx6q14.c:81:31: fatal error: linux/busfreq-imx.h: No such file or directory
|  #include <linux/busfreq-imx.h>
|                                ^
| compilation terminated.
```

fmntf, 2017-02-01:
Please try to build this branch: https://github.com/fmntf/linux_kernel/tree/4.1.15_2.x-udoo
It contains fixes for GCC 6

    make mrproper
    ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make udoo_neo_defconfig
    ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- make imx6sx-udoo-neo-{basic,basicks,extended,full}{-hdmi,-lvds7,-lvds15,}{-m4,}.dtb zImage modules firmware -j8

cyrilfr, 2017-02-01:
It doesn't compile that way as the default udoo_neo_defconfig file enable ttyMCC which is not supported anymore. Otherwise, it compiles. There is probably an issue with the recipe or the OpenEmbedded environment. I'll have a look on this.

fmntf, 2017-02-01:
Sorry, in the directory where I have built 4.1.15-2.0.0 I had some uncommitted changes (an initial draft of ttyMCC on 4.1). Btw, if you disable ttyMCC from defconfig you should be able to build a fully working kernel. SD card should work too.

cyrilfr, 2017-02-01:
Well, compiling with BitBake (in OpenEmbedded) still doesn't work for me. Here is the most important part of the logs :

```
cat: drivers/net/wireless/ti/wlcore/release_version.h: No such file or directory
| /home/eeproperty/Documents/Vesta/VestaOS/poky/build-neo-morty/tmp/work-shared/udooneo/kernel-source/drivers/net/wireless/ti/wlcore/Makefile:38: recipe for target 'drivers/net/wireless/ti/wlcore/version.h' failed
| make[7]: *** [drivers/net/wireless/ti/wlcore/version.h] Error 1
| /home/eeproperty/Documents/Vesta/VestaOS/poky/build-neo-morty/tmp/work-shared/udooneo/kernel-source/scripts/Makefile.build:403: recipe for target 'drivers/net/wireless/ti/wlcore' failed
| make[6]: *** [drivers/net/wireless/ti/wlcore] Error 2
| make[6]: *** Waiting for unfinished jobs....
```
But surprisingly (or not), the file `drivers/net/wireless/ti/wlcore/release_version.h` DOES exist. I'll have to figure out why the `cat` command in the Makefile doesn't find it :/

graugans, 2017-02-01:
@cyrilfr This maybe caused by an issue with out of source builds. You can try to set ``B=${S}`` in your bbappend file. This may break the building of the vivante driver afterwards.

cyrilfr, 2017-02-01:
@graugans That was the issue! I do compile all the branches available now. None of them boot correctly. The symptom is still a wait on root device /dev/mmcblkop1. Maybe an error in the WIC script?

graugans, 2017-02-01:
No, wic is just for creating the image. I guess the issue is kernel related. I'll check on Friday or Saturday.

JochenLoeser, 2017-03-31:
I have the same problem.
I don't have a real solution but after verifing, that the sd card has power I looked through the kernel source code.
Apparently the sd card driver does not work with my sd card.
I've got a https://www.google.com/search?q=sandisk+ultra+32gb+sdhc+micro+sd (at least the picture is the same)
I took the identical image with a 'normal' non SDHC card and it worked!

I tracked the -110 error to the function "mmc_sd_get_cid".
Sidenote: The card works with 4.10 mainline and the 3.14 from udoo.

I hope this helps the smarter more experienced people to find a solution.
If I can provide any further help or anything feel free to contact me.

modjo756, 2017-04-01:
Hi all, this week i had a formation on yocto and linux embedded with phytec in France. During the training we use there mir3 board (imx6 quad like udoo quad and seco sbc a62). Phytec are on linux mainline for [linux-kernel](https://git.phytec.de/linux-mainline/?h=v4.1.18-phy) (very interesting for maintaining and update !). Perhaps we could have a look on their repository to check if we can adapt something to our card ...) . Otherwise they use barebox for boot loader. Actually i'm trying to make a meta for the openZwave library, i let you know on udoo forum ...

JochenLoeser, 2017-04-05:
I think I found a(the?) solution: `no-1-8-v;` is missing. See: https://github.com/UDOOboard/linux_kernel/blob/3.14-1.0.x-udoo/arch/arm/boot/dts/imx6sx-udoo-neo.dtsi#L529
As far as I understand the Device Tree (and please correct me if I'm wrong - I want to learn something): The sdcard interface (usdhc2) is supplied by regulator vcc_sd2. This is a fixed regulator (the min/max values states this too). This means the kernel will never be able to apply 1.8V.

On the other hand shouldn't the kernel be able to figure this out?

graugans, 2017-04-05:
@JochenLoeser Have you checked the schematics how the 1.8v is generated? I have so much i.mx6 boards floating arround not sure how this is done in UDOO at the moment. The frequency and voltages is still a mysterium in device tree for me as-well. I guess the handling of regulator have changed in 4.x kernels.

JochenLoeser, 2017-04-05:
My first thought (before I even found this issue) was that the inversion of the chip-detect pin ( https://github.com/fmntf/linux_kernel/commit/0dffbfaa5b5877ec32bd31c709eefe1e8c713358#diff-c63642f2e8268f6942b422d66eeda0e0L356 ) was a mistake. So I measured the supply voltage at the sdcard connector: 3.3V

I have to double check with the real board but according to the schematics the sdcard VDD is the same net as the Arduino power supply on the pinheaders. This should be fixed at 3.3V. It would be weird, if the sdcard can say at which voltage the Arduino is running.

But as I said I will double check tomorrow.

fmntf, 2017-04-06:
Hi guys. `no-1-8-v;` was removed by mistake. I will bring it back, thank you for noticing!
The SD card runs at 3.3V only. You can verify it on the Neo revD schematics on page 8. `NVCC_3V0` and `VCC_SW` are fixed and cannot be changed.

The card detect pin logic was probably not checked on Linux 3.14, where it was wrongly declared as GPIO_ACTIVE_HIGH instead of GPIO_ACTIVE_LOW and everything worked. On 4.1, instead, the SD was not detected until that fix. That fix should be fixed on 3.14 too, but since everything works there...

modjo756, 2017-04-06:
Hi all, i saw they have some progress on kernel 4.1 ... good news ! Graugans, do you plan to integrate this kernel on morty branch ?

fmntf, 2017-04-06:
Remember there is no M4 communication support on this kernel. I don't even know if M4 can boot at all, I did not verify the shared memory pools.

modjo756, 2017-04-06:
thanks for this precision @fmntf , could you explain (rapidly) why this works on 3.14 and not in 4.1 please ? where is your driver that enable this on 3.14 ?

cyrilfr, 2017-04-07:
@fmntf for people like me who don't use the M4, moving forward to 4.1 and beyond worth it.

fmntf, 2017-04-07:
Yeah, I know. But since it is not complete, we cannot move it on the official UDOO repo.
If you are interested there is an Ubuntu 16.04 image with this kernel and most of the standard UDOO stuff (GPU acceleration on Xorg and QT, ...) [here](https://drive.google.com/open?id=0B400VKg9Vc0rVkEtS1hOTWVFYjA).
Regarding the missing M4 stuff: NXP deprecated the MCC stuff just after 3.14, they use rpmsg now. We need to bind Arduino calls to rpmsg instead of mcc. However I have no skills on that, and the guys able to do that on our team are really busy. There is no eta. I have also tried to port MCC to 4.1 as a workarout, but I do not wanna go to the hell :D

cyrilfr, 2017-04-20:
@modjo756 Sorry I moved back temporary to 3.14 because I had to fix other issues, especially the uEnv.txt file that isn't copied in the boot/ directory. Patches probably don't work anymore with 4.1.15.

modjo756, 2017-04-20:
Ok thanks, if i understand, you didn't use Francesco implementation of kernel ?

cyrilfr, 2017-04-20:
No I didn't.

modjo756, 2017-04-20:
Ok i just build a yocto image with @graugans layer (morty branch) and kernel 4.1.15 from @fmntf . It compile but it's needed to remove ttyMCC module and i needed to add this patch : 

`diff --git a/drivers/input/misc/fxas2100x.c b/drivers/input/misc/fxas2100x.c
index 3984dbb..bec0bd2 100644
--- a/drivers/input/misc/fxas2100x.c
+++ b/drivers/input/misc/fxas2100x.c
@@ -457,13 +457,13 @@ static ssize_t fxas2100x_data_show(struct device *dev,
 }
 
 
-static DEVICE_ATTR(enable, 0666, fxas2100x_enable_show, fxas2100x_enable_store);
+static DEVICE_ATTR(enable, 0660, fxas2100x_enable_show, fxas2100x_enable_store);
 
-static DEVICE_ATTR(poll_delay, 0666,fxas2100x_poll_delay_show, fxas2100x_poll_delay_store);
+static DEVICE_ATTR(poll_delay, 0660,fxas2100x_poll_delay_show, fxas2100x_poll_delay_store);
 
-static DEVICE_ATTR(position, 0666,fxas2100x_position_show, fxas2100x_position_store);
+static DEVICE_ATTR(position, 0660,fxas2100x_position_show, fxas2100x_position_store);
 
-static DEVICE_ATTR(data, 0666,fxas2100x_data_show, NULL);
+static DEVICE_ATTR(data, 0660,fxas2100x_data_show, NULL);
 
 
 static struct attribute *fxas2100x_attributes[] = {
diff --git a/drivers/input/misc/fxos8700.c b/drivers/input/misc/fxos8700.c
index 3394937..d63c440 100644
--- a/drivers/input/misc/fxos8700.c
+++ b/drivers/input/misc/fxos8700.c
@@ -592,13 +592,13 @@ static ssize_t fxos8700_data_show(struct device *dev,
 	return sprintf(buf, "%d,%d,%d\n",data.x,data.y,data.z);
 }
 
-static DEVICE_ATTR(enable, 0666,fxos8700_enable_show, fxos8700_enable_store);
+static DEVICE_ATTR(enable, 0660,fxos8700_enable_show, fxos8700_enable_store);
 
-static DEVICE_ATTR(poll_delay,0666,fxos8700_poll_delay_show, fxos8700_poll_delay_store);
+static DEVICE_ATTR(poll_delay,0660,fxos8700_poll_delay_show, fxos8700_poll_delay_store);
 
-static DEVICE_ATTR(position, 0666,fxos8700_position_show, fxos8700_position_store);
+static DEVICE_ATTR(position, 0660,fxos8700_position_show, fxos8700_position_store);
 
-static DEVICE_ATTR(data, 0666,fxos8700_data_show, NULL);
+static DEVICE_ATTR(data, 0660,fxos8700_data_show, NULL);
 
 
 /*acc only sysfs interface*/
@@ -625,7 +625,7 @@ static ssize_t fxos8700_motion_detect_store(struct device *dev,
 	return count;
 }
 
-static DEVICE_ATTR(motion_detect, 0666,fxos8700_motion_detect_show, fxos8700_motion_detect_store);
+static DEVICE_ATTR(motion_detect, 0660,fxos8700_motion_detect_show, fxos8700_motion_detect_store);
 
 static struct attribute *fxos8700_acc_attributes[] = {
 	&dev_attr_enable.attr,`

 to remove error during the build 
`include/linux/bug.h:33:45: error: negative width in bit-field ‘<anonymous>’
 #define BUILD_BUG_ON_ZERO(e) (sizeof(struct { int:-!!(e); }))`

Now the card boot fine : 

`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.15_2.0.x-udoo+g7773e46 (mls@KDE-Z400) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT Thu Apr 20 10:58:07 CEST 2017
[    0.000000] CPU: ARMv7 Processor [412fc09a] revision 10 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: UDOO Neo Full`

but have issues with CAN Bus (like kernel 3.14.56 ...)

cyrilfr, 2017-04-20:
How did you manage to put the uEnv.txt file on the SD card?

modjo756, 2017-04-20:
I use @graugans layers (Morty) didn't make modifs for this point

cyrilfr, 2017-04-20:
So this doesn't work with LVDS screen?

modjo756, 2017-04-20:
I didn't test lvds screen for the moment (i use HDMI) ...

cyrilfr, 2017-04-23:
@graugans, @modjo756 did you manage to use an LVDS screen with the meta-freescale u-boot recipe?

modjo756, 2017-04-24:
@cyrilfr , if you talk about the u-boot used in graugans layers, yes i use lvds screen with all the card fine (neo, quad and sa62) --> krogoth Branch and kernel 3.14.56 ....

cyrilfr, 2017-04-24:
How does it load the uEnv.txt file in the "boot/" directory? In my case it doesn't.

modjo756, 2017-04-24:
Yes the uEnv file is in the boot/ folder

modjo756, 2017-04-30:
@cyrilfr  you are right about the problem to load uEnv.txt (LVDS output) i have the same issue, this is due that at boot stage (u-boot) the file system is not recognise ! see under : 

`U-Boot SPL 2016.11+fslc+gc44711d (Apr 30 2017 - 22:36:41)
Trying to boot from MMC1

U-Boot 2016.11+fslc+gc44711d (Apr 30 2017 - 22:36:41 +0200)

CPU:   Freescale i.MX6Q rev1.2 at 792 MHz
Reset cause: POR
Board: UDOO Quad
DRAM:  1 GiB
MMC:   FSL_SDHC: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   using phy at 6
FEC [PRIME]
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
** Unrecognized filesystem type **
Device Tree: /boot/imx6q-udoo-lvds15.dtb
4534496 bytes read in 491 ms (8.8 MiB/s)
Booting from mmc ...
44202 bytes read in 216 ms (199.2 KiB/s)
## Flattened Device Tree blob at 18000000
   Booting using the fdt blob at 0x18000000
   Using Device Tree in place at 18000000, end 1800dca9

Starting kernel ...
`

we can see : Unrecognized filesystem type ...

Perhaps a problem with wic, @graugans if you can look at this  (for information i have the lvds output selected because i modify the default selection in udooinit function)

Otherwise, @fmntf  i can boot with kernel 4.1.15 on udoo quad, with eth working fine, arduino ... Qt 5.8 and morty branch !

cyrilfr, 2017-05-01:
No there is not issue with the filesystem. If you copy the uEnvt.txt file in the boot/ directory manually, it will work. We just need to put this file at the right location inside the bbappend file. If you look at your u-boot-fsl.bbappend file, nothing put the uEnv.txt file in boot/.

modjo756, 2017-05-01:
yes but i have an uEnv.txt in /boot --> see uenv_0.1.bb in recipes-udoo/environment and there is the same file in /recipes-bsp/u-boot/u-boot-fslc/udooqdl(or udooneo) ?!

cyrilfr, 2017-05-02:
OK but why using an external recipe instead of using the u-boot bbappend file ?

modjo756, 2017-05-02:
I don't know, only @graugans can answer on this point !

graugans, 2017-05-03:
@cyrilfr I tried this approach but as far as I remember this caused an issue because u-boot is not intended to be installed on the image. When I got rid of the ``/boot`` partition and read the ``uEnv.txt`` file directly from the rootfs either a major rewrite of the bbappend or a new recipe was needed. I decided to write a new recipe. Skipping the ``/boot`` partition makes things more reliable in my eyes because I do not trust the FAT filesystem.

graugans, 2017-05-03:
@modjo756 and @cyrilfr It looks like both of you have recipes for kernel 4.1 at least on your local machine. It would be very nice to see some Patches or even Pull-Request. If needed I add an separate branch so we can try to fix things up. I am very short in time but would be fun to get 4.1 kernel up and running.

cyrilfr, 2017-05-03:
@graugans I was stucked to boot the kernel but once I'll get it work, probably next week, I'll detail every single step and push all the recipes.

modjo756, 2017-05-03:
@graugans see [here ](https://github.com/graugans/meta-udoo/issues/34#issuecomment-296555578)for my kernel 4.1 (first commit), i will update it this week. It will be great if you can push a dev branch on morty kernel 4.1 !

graugans, 2017-05-03:
I just pushed a preliminary recipe for [kernel 4.1](https://github.com/graugans/meta-udoo/tree/morty-4.1) based on the work from @modjo756

modjo756, 2017-05-03:
This evening, trying to boot kernel 4.1 on secoA62 but some problem due to   ** Unrecognized filesystem type ** at boot stage ... tried also to load image from tftp with no success ! We need to resolve the problem with /boot

graugans, 2017-05-04:
I guess this is again related to the on board emmc and the different ordering of sdcard on A62

modjo756, 2017-05-04:
Yes i suppose, i have tried with different options (env u-boot) but no success !

modjo756, 2017-06-29:
For information seco as provided a a bsp for it's board with kernel 4.1.15 here and [here](https://secogit.seco.com/imx6_release/linux-4-1-15-secoboards-imx6-rel/tree/master) and here the [bsp](https://secogit.seco.com/imx6_release/meta-seco-bsp-release-2-0-rel/tree/master). i will try and let you know how it works !

graugans, 2017-06-30:
Thanks for the hint.

modjo756, 2017-07-30:
@graugans @drewmoseley , with my qdl i use kernel 4.1 that works fine but not for wifi. At first boot i have it i can connect to it, but at second boot i did'nt have wlan0, see the error at boot : 
Configuring network interfaces... [    4.291123] usb 1-1.3: New USB device found, idVendor=148f, idProduct=5370
[    4.298107] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.298116] usb 1-1.3: Product: 802.11 n WLAN
[    4.298122] usb 1-1.3: Manufacturer: Ralink
[    4.298127] usb 1-1.3: SerialNumber: 1.0
[    4.393115] usb 1-1.4: new full-speed USB device number 5 using ci_hdrc
[    4.530610] usb 1-1.4: New USB device found, idVendor=0403, idProduct=6001
[    4.537558] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.544923] usb 1-1.4: Product: EnOcean USB 300 DB
[    4.549723] usb 1-1.4: Manufacturer: EnOcean GmbH
[    4.554454] usb 1-1.4: SerialNumber: FTXH4VLL
Successfully initialized wpa_supplicant
[    4.579402] usbcore: registered new interface driver usbserial
[    4.585668] usbcore: registered new interface driver usbserial_generic
[    4.592293] usbserial: USB Serial support registered for generic
[    4.603106] usb 1-1.3: reset high-speed USB device number 4 using ci_hdrc
[    4.608281] usbcore: registered new interface driver ftdi_sio
[    4.609022] usbserial: USB Serial support registered for FTDI USB Serial Device
[    4.609575] ftdi_sio 1-1.4:1.0: FTDI USB Serial Device converter detected
[    4.609730] usb 1-1.4: Detected FT232RL
[    4.610444] usb 1-1.4: FTDI USB Serial Device converter now attached to ttyUSB0
rfkill: Cannot open RFKILL control device
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface
wlan0: ERROR while getting interface flags: No such device
SIOCSIFADDR: No such device
wlan0: ERROR while getting interface flags: No such device
SIOCSIFNETMASK: No such device
SIOCADDRT: No such device
[    4.764223] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 5390, rev 0502 detected
[    4.802226] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 5370 detected
[    4.884176] fec 2188000.ethernet eth0: Freescale FEC PHY driver [Generic PHY] (mii_bus:phy_addr=2188000.ethernet:06, irq=-1)
[    4.899278] usbcore: registered new interface driver rt2800usb
Starting system message bus: dbus.
Starting OpenBSD Secure Shell server: sshd
[    5.277762] NET: Registered protocol family 10
[    5.283776] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready

Do you have an idea ?

drewmoseley, 2017-07-30:
No clue, sorry.  I only have tried it with wired Ethernet.

modjo756, 2017-07-30:
Tried with kernel 3.14 but same issue ... perhap's a problem with samba, apache , ftp ...

cyrilfr, 2017-08-22:
@graugans I tried to compile an image with your kernel 4.1 branch. It seems to work fine except that the LVDS 7'' touchscreen events aren't triggered. I tried event0, event1 and event2 as input in the qt-env.sh file.

[38 more replies]