graugans/meta-udoo #19: yocto don't boot on secoA62 emmc

modjo756, 2016-10-10:
For the moment yocto don't boot on secoA62 emmc. Apparently a problem from uboot/SPL that is not the good version for the board. See [here](http://www.udoo.org/forum/threads/copy-yocto-image-to-emmc.4969/) for more details

graugans, 2016-10-10:
This is due to the fact the internal emmc is not yet activated

graugans, 2016-10-10:
I have a preliminary patch available to activate eMMC in u-boot. At the moment the the devices are in the wrong order.  U-Boot now tries to boot from emmc when booted from SD-Card... When this is sorted out I try booting from eMMC.

graugans, 2016-10-11:
After [changing the order](https://github.com/graugans/meta-udoo/blob/65f09f00e9792a1b5e5e08c3e232ee7aeeeced64/recipes-bsp/u-boot/u-boot-seco/secosbca62/0001-Added-preliminary-eMMC-support-for-Seco-A62.patch) the board boots at least from uSD card but eMMC still fails, because the loading of the kernel tries to load from uSD. This needs more investigation

```
U-Boot SPL 2016.01-seco-a62-v2016.01+g654be7a (Oct 11 2016 - 22:20:50)
Trying to boot from MMC
mmc port 3

U-Boot 2016.01-seco-a62-v2016.01+g654be7a (Oct 11 2016 - 22:20:50 +0200)

CPU:   Freescale i.MX6DL rev1.1 at 792 MHz
Reset cause: POR
Board: Seco SBC-A62-J-LITE
       Watchdog enabled
I2C:   ready
DRAM:  1 GiB
MMC:   FSL_SDHC: 0, FSL_SDHC: 1
MMC: no card present
*** Warning - MMC init failed, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   using phy at 6
FEC [PRIME]
Hit any key to stop autoboot:  0 
MMC: no card present
MMC: no card present
Booting from net ...
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
*** Unhandled DHCP Option in OFFER/ACK: 42
*** Unhandled DHCP Option in OFFER/ACK: 158
*** Unhandled DHCP Option in OFFER/ACK: 42
*** Unhandled DHCP Option in OFFER/ACK: 158
DHCP client bound to address 192.168.23.174 (1029 ms)
Using FEC device
TFTP from server 192.168.23.1; our IP address is 192.168.23.174
Filename 'zImage'.
Load address: 0x12000000
Loading: *
```

modjo756, 2016-10-11:
Thanks for the details of your job !

graugans, 2016-10-12:
I just updated the patch. If you want to test the latest image you can download a command line only  image [here](https://buildr.ege.io/YOCTO/krogoth/udoo/udoo-image-full-cmdline-secosbca62.sdcard.gz)

modjo756, 2016-10-12:
Nice work ! tested and approved :+1:  the card boot on emmc when i removed the jumper (just resizing script not work apparently).  If i replug the jumper, the card boot on sd !

graugans, 2016-10-13:
Another topic might be the u-boot environment location. But this is low prio I think because the way to go is using uEnv.txt in /boot partition

graugans, 2016-10-17:
I did some test yesterday based on the resize script. I have to rework the script to use the kernel command line to detect the rootfs partition relying on the disk label does not work because sometimes I get the label from the uSD card.

modjo756, 2016-11-17:
After a test with u-boot-fslc the card boot on emmc but for that i need at boot : 
_setenv mmcroot /dev/mmcblk0p1
saveenv_
after the card boot on emmc and your resize script work fine. But if i remove the sdcard, i have this error : 
_U-Boot SPL 2016.07+fslc+ge6b4241 (Nov 15 2016 - 17:35:18)
Trying to boot from MMC1
Card did not respond to voltage select!
spl: mmc init failed with error: -17
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###_
It would be great to boot on emmc without sdcard.

graugans, 2016-11-17:
Yes, this is caused due to the different logic of boot selection. I have to check why u-boot fails to boot from emmc....

graugans, 2016-11-20:
This was fixed with commit [30a6fc0e74c7971fc7bd8457f5d64528a8133079](https://github.com/graugans/meta-udoo/commit/30a6fc0e74c7971fc7bd8457f5d64528a8133079)

closed 2016-11-20 by graugans

modjo756, 2016-11-20:
Good, i will try tomorrow and let you known.

modjo756, 2016-11-21:
tested this morning and work fine. Nice job !