graugans/meta-udoo #20: sdk package for A62 not working (qt)

modjo756, 2016-10-11:
I have build a sdk package for sa62 (build with qt5.6.2, meta-qt5 layer krogoth branch). The sdk build fine. After install the package and configure qt creator for the kit, i have this error when i try to deploy an app : 

`/usr/bin/ld: cannot find -lQt5Widgets
/usr/bin/ld: cannot find -lQt5Gui
/usr/bin/ld: cannot find -lQt5SerialPort
/usr/bin/ld: cannot find -lQt5Core
/usr/bin/ld: cannot find -lGLESv2
collect2: error: ld returned 1 exit status
make: *** [terminal] Error 1`
[image]

if i use the neo kit (qt 5.6.1) that i already have the app compile and i can deploy

if i compare the qt5 details of this 2 conf, i can see a difference with mkspec : 
neo : linux-oe-g++
[image]

secoA62 : linux-g++

[image]

i don't know if is the problem ...

graugans, 2016-10-12:
I'll dig into this when the [eMMC boot](https://github.com/graugans/meta-udoo/issues/19) works

modjo756, 2016-10-13:
After a test with udoo quad, same error --> not working. I don't know if it's an issue with meta qt5 or meta-oe ?! I will try a build with an older meta-qt5 to see ...

graugans, 2016-10-14:
I'll take a look into this at the weekend

cyrilfr, 2016-10-16:
Sounds like a linking issue. Did you add the modules to your .pro file?

modjo756, 2016-10-16:
Yes, this is not the problem (i use project that works before with the older version qt5.6 or example). I think it's a problem with mkspec (use linux-g++ but we needs linux-oe-g++) ! see this [comparison](https://cloud.githubusercontent.com/assets/5494427/19353561/0c2e1a00-9164-11e6-82c4-c21de0b07266.png) between an older qmake that worked with the new generated.

cyrilfr, 2016-10-16:
Personally I use that layer : [https://github.com/meta-qt5/meta-qt5](https://github.com/meta-qt5/meta-qt5). It has nothing to do with meta-freescale.

modjo756, 2016-10-16:
Yes i use also meta-qt5, but on meta-freescale they have otavio salvador who is a maintainer of meta-qt5. If you know a mailing list, forum ... on meta-qt5 let me know please.

modjo756, 2016-10-18:
@graugans @cyrilfr i have found the problem with sdk. Now to have the deploy working, it's needed to add linux-oe-g++ to Qt mkspec, see the picture under !
[image]

graugans, 2016-10-19:
Hi, what branch are you using? For krogoth a krogoth branch in meta-qt5 is recommended. It could be possible names of tools changed and so the qt5 tutorial needs an update. If you find some time to create a minimal example we can provide this and refer to it

modjo756, 2016-10-19:
Hi graugans, i use master for meta-qt5 and krogoth for the other (but this is the same result with krogoth branch, tested). The only thing that is needed to change is to add linux-oe-g++ to Qt mkspes field.  I can create an example but can you preceise your ask ? Would you like a tutorial to configure QtCreator or deployment  example ?

modjo756, 2016-10-20:
I close this issue, another user have tested it (mailing list meta-freescale)  and works too for him.

closed 2016-10-20 by modjo756

modjo756, 2016-11-17:
@graugans , @cyrilfr about this problem i posted on yocto mailing list this issue. Someone post me a private message that explain this : 
_Problem lays in the patch 0008-configure-paths-for-target-qmake-properly.patch used in nativesdk-qtbase_git.bb.

From the patch:
+if [ "$QT_CROSS_COMPILE" = "yes" ] ; then
+ shortxspec=linux-g++
+ shortspec=linux-g++
+else
+ shortxspec=`echo $XQMAKESPEC | sed "s,^${relpath}/mkspecs/,,"`
+ shortspec=`echo $QMAKESPEC | sed "s,^${relpath}/mkspecs/,,"`
+fi

Since QT_CROSS_COMPILE then it sets the SPEC to linux-g++.

To test if you have the same problem as me.

Remove the patch from the recipe nativesdk-qtbase_git.bb and rebuild.

diff --git a/recipes-qt/qt5/nativesdk-qtbase_git.bb b/recipes-qt/qt5/nativesdk-qtbase_git.bb
index 7452648..a941c1f 100644
--- a/recipes-qt/qt5/nativesdk-qtbase_git.bb
+++ b/recipes-qt/qt5/nativesdk-qtbase_git.bb
@@ -33,7 +33,6 @@ SRC_URI += "\
 file://0005-configure-bump-path-length-from-256-to-512-character.patch \
 file://0006-QOpenGLPaintDevice-sub-area-support.patch \
 file://0007-linux-oe-g-Invert-conditional-for-defining-QT_SOCKLE.patch \
- file://0008-configure-paths-for-target-qmake-properly.patch \
 file://0009-Reorder-EGL-libraries-from-pkgconfig-and-defaults.patch \
 file://0010-Pretend-Qt5-wasn-t-found-if-OE_QMAKE_PATH_EXTERNAL_H.patch \_

In my last build i did this and this resolve the problem (no need to change QT mkspecs in qtcreator). Do you think we push this on the recipe ?