graugans/meta-udoo #28: ethernet not working on seco A62

modjo756, 2016-11-11:
With the new build eth is not working on secoA62, log at boot give this error 

_Configuring network interfaces... [    4.182171] fec 2188000.ethernet eth0: no PHY, assuming direct connection to switch
[    4.189884] libphy: PHY fixed-0:00 not found
[    4.194210] fec 2188000.ethernet eth0: could not attach to PHY
SIOCSIFFLAGS: No such device

[    4.214395] fec 2188000.ethernet eth0: no PHY, assuming direct connection to switch
[    4.222065] libphy: PHY fixed-0:00 not found
[    4.226354] fec 2188000.ethernet eth0: could not attach to PHY
SIOCSIFFLAGS: No such device_

see the full log [here](https://gist.github.com/modjo756/5cc9761d653862d3cf2fcae5371c04a6)

graugans, 2016-11-14:
From a quick check it looks like the device tree entry is looking different:

[Old kernel](https://github.com/UDOOboard/linux_kernel/blob/imx_3.14.28_1.0.0_ga_a62/arch/arm/boot/dts/imx6qdl-seco_SBC_A62.dtsi#L292) vs. [new Kernel](https://github.com/UDOOboard/linux_kernel/blob/imx_3.14.28_1.0.0_ga_a62/arch/arm/boot/dts/imx6qdl-seco_SBC_A62.dtsi#L292)

OLD (working) one
```
		pinctrl_enet: enetgrp {
			fsl,pins = <
				MX6QDL_PAD_ENET_MDIO__ENET_MDIO              0x1b0b0
				MX6QDL_PAD_ENET_MDC__ENET_MDC                0x1b0b0
				MX6QDL_PAD_RGMII_RD0__RGMII_RD0              0x1b0b0
				MX6QDL_PAD_RGMII_RD1__RGMII_RD1              0x1b0b0
				MX6QDL_PAD_RGMII_RD2__RGMII_RD2              0x1b0b0
				MX6QDL_PAD_RGMII_RD3__RGMII_RD3              0x1b0b0
				MX6QDL_PAD_RGMII_RX_CTL__RGMII_RX_CTL        0x1b0b0
				MX6QDL_PAD_RGMII_TD0__RGMII_TD0              0x1b0b0
				MX6QDL_PAD_RGMII_TD1__RGMII_TD1              0x1b0b0
				MX6QDL_PAD_RGMII_TD2__RGMII_TD2              0x1b0b0
				MX6QDL_PAD_RGMII_TD3__RGMII_TD3              0x1b0b0
				MX6QDL_PAD_RGMII_TX_CTL__RGMII_TX_CTL        0x1b0b0
				MX6QDL_PAD_RGMII_RXC__RGMII_RXC              0x1b0b0
				MX6QDL_PAD_RGMII_TXC__RGMII_TXC	             0x1b0b0
				MX6QDL_PAD_ENET_REF_CLK__ENET_TX_CLK         0x1b0b0

				/*  Phy Interrupt  */
				MX6QDL_PAD_ENET_TX_EN__GPIO1_IO28            0x1f071
				/*  ENET RESET */
				MX6QDL_PAD_EIM_D23__GPIO3_IO23               0x1f071
			>;
		};
```

New (not working) one
```
		pinctrl_enet: enetgrp {
			fsl,pins = <
				MX6QDL_PAD_ENET_MDIO__ENET_MDIO		0x1b0b0
				MX6QDL_PAD_ENET_MDC__ENET_MDC		0x1b0b0
				MX6QDL_PAD_RGMII_RD0__RGMII_RD0		0x1b0b0
				MX6QDL_PAD_RGMII_RD1__RGMII_RD1		0x1b0b0
				MX6QDL_PAD_RGMII_RD2__RGMII_RD2		0x1b0b0
				MX6QDL_PAD_RGMII_RD3__RGMII_RD3		0x1b0b0
				MX6QDL_PAD_RGMII_TD0__RGMII_TD0		0x1b0b0
				MX6QDL_PAD_RGMII_TD1__RGMII_TD1		0x1b0b0
				MX6QDL_PAD_RGMII_TD2__RGMII_TD2		0x1b0b0
				MX6QDL_PAD_RGMII_TD3__RGMII_TD3		0x1b0b0
				MX6QDL_PAD_RGMII_TX_CTL__RGMII_TX_CTL	0x1b0b0
				MX6QDL_PAD_RGMII_RXC__RGMII_RXC		0x1b0b0
				MX6QDL_PAD_RGMII_TXC__RGMII_TXC		0x1b0b0
				MX6QDL_PAD_ENET_REF_CLK__ENET_TX_CLK	0x1b0b0
				MX6QDL_PAD_EIM_D23__GPIO3_IO23          0x80000000
			>;
		};

		phy_running: phy_runningrp {
			fsl,pins = <
				MX6QDL_PAD_RGMII_RX_CTL__RGMII_RX_CTL	0x1b0b0
			>;
		};
      
		phy_reset: phy_resetgrp {
			fsl,pins = <
				MX6QDL_PAD_RGMII_RX_CTL__GPIO6_IO24	0x80000000
			>;
		};
```

graugans, 2016-11-14:
Maybe one of the guys from Seco/UDOO can assist here.
- @DavideCardillo
- @Ekironji
- @fmntf

I am not a device tree pro.

stefano-garzarella, 2016-11-14:
I confirm, I have exactly the same problem!

modjo756, 2016-11-14:
I make some research (on google) about this problem this morning and find nothing to help ! This afternoon i will try a build with a patch to come back to old kernel version (only for eth pin) and check if it works ...

graugans, 2016-11-14:
@stefano-garzarella what buildsystem/distributio are you using?

For more details about the device tree binding can be found [here](https://github.com/UDOOboard/linux_kernel/blob/3.14-1.0.x-udoo/Documentation/devicetree/bindings/net/fsl-fec.txt)

graugans, 2016-11-15:
Please check with the latest commit. For me it is working within a Gigabit network

modjo756, 2016-11-15:
@graugans , i think your commit works, because i did the same today and it also working for me, see [here](https://github.com/modjo756/meta-udoo-modjo/commit/b674c97227bf3e915c2df4ba734171776d7252cd). In this patch i also added can,spi,i2c ... 
I also added a secosbca62 folder with uEnv.txt in uboot to build uboot for A62 (if not i need to build it manually !)

graugans, 2016-11-15:
Okay, I'll check my uEnv.txt logic tomorrow. I recommend to split the patch for CAN, I2C and SPI. Maybe we can push the Ethernet changes upstream to the UDOO repo.

fmntf, 2016-11-15:
Yes, pull requests are welcome!

modjo756, 2016-11-17:
I have split my patch in some modules, see [here](https://github.com/modjo756/meta-udoo-modjo/commit/1ca20d0d10a62a4e24705fdabf1b01935ab838ce). Network is ok now, i close this issue.

closed 2016-11-17 by modjo756