graugans/meta-udoo #30: upload scketch to udooneo not working

modjo756, 2016-12-06:
Trying to load sketch from arduino ide (from host pc) over network not working (take long times to upload without errors but nothing change on the card, not update of M4 firmware) . If i'm connect to the card and see /var/log/udoofota-server.log i see this : 

_root@udooneo:/var/log# cat udoofota-server.log 
[2016-12-05 22:34:59 UTC] (sys) Starting_

it seems to be ok, and on znmap i can see that port 5152 is open. After that i kill the service and restart it. On pc side relaunch the upload of the sketch, after some times i get this error : 

_root@udooneo:/var/log# udooserver 
Connection accepted on 5152 port
Uploader command received
Received fileSize = 110601
Command = udooneo-m4uploader M4_file.fw
/usr/bin/udooneo-m4uploader: line 93:   522 Bus error               $up "$INPUT"
flash: uploading input firmware failed
status = 34560
Remote Command [2] done
Close connection on 5152 port_

Apparently an error with udooneo-m4uploader. Same procedure tested on udoobuntu2.1.1 give me a good result (upload sketch ok)

modjo756, 2016-12-07:
After some test, it seems that for yocto we need to use udooclient (udoofota) V1.0 and not V1.2 to load the code on the card. this work with the precedent build (kernel 3.14.28 and uboot udoo) but not with the last build 

root@udooneo:~# udooserver 
Connection accepted on 5152 port
Uploader command received
Received fileSize = 110601
Command = udooneo-m4uploader M4_file.fw
UDOONeo - Failed to Start M4 sketch: reboot system ! 
flash: uploading input firmware failed
status = 768
Remote Command [2] done
Close connection on 5152 port

cyrilfr, 2016-12-08:
Did you move forward to the last commits of udoofota and m4uploader?

modjo756, 2016-12-08:
yes you are right, we use an older version of this files ... I will update it and try a new build !

modjo756, 2016-12-08:
After a test (change SRCREV in .bb to lasted rev), we can use udooclientV1.2 but we yet have an error on the card : 
_root@udooneo:~# udooserver 
Listening connections on port 5152
Connection accepted on port 5152
bytesReceived = 13
Uploader command received
Received end of file
Received fileSize = 110548
Command = udooneo-m4uploader M4_file.fw
/usr/bin/udooneo-m4uploader: line 93:   444 Bus error               $up "$INPUT"
flash: uploading input firmware failed
statusUploader = 135
Remote Command [2] done
bytesReceived = 21
GET_M4_STATUS received
Remote Command [3] done
bytesReceived = 0
Closed connection on port 5152_

perhaps @fmntf  can help us on it ...

fmntf, 2016-12-09:
Hi; don't know.. that script should not be used. I directly use mqx_upload_on_m4SoloX. In fact, in Android there is only mqx_upload_on_m4SoloX and not udooneo-m4uploader.

modjo756, 2016-12-09:
@graugans , how we can modify the layer to use mqx_upload_on_m4SoloX instead of udooneo_m4uploader ?

fmntf, 2016-12-09:
In [udooserver.c](https://github.com/UDOOboard/udoofota/blob/master/udooserver.c) `M4_FIRMWARE_RECEIVED_FILE` should become `/var/opt/m4/m4last.fw`. Then `mqx_upload_on_m4SoloX` can be symlinked to `udooneo-m4uploader`, so the script will not be present at all in the rootfs.

That's how it is implemented on Android, by direct suggestion of the original `mqx_upload_on_m4SoloX` authors from Seco.

modjo756, 2016-12-09:
make a patch to change M4_FIRMWARE_RECEIVED_FILE and works perfect ! i need to make some change before to pull the patch and close the issue. Thanks for your help fmntf

modjo756, 2016-12-09:
Add a pull request to correct this issue. Only one problem persist. At first boot we have the same error but after a reset we can upload the sketch ... perhaps a problem with the init script ?