graugans/meta-udoo #33: No fb0 on SBCA62 solo

modjo756, 2016-12-16:
Hi, i'm trying to use a scosbcA62 solo with yocto image but i have a problem with frame buffer. I want to use lvds with it but i didn't have fb0 in /dev ? see [here](https://gist.github.com/modjo756/8caf8379b1ad21e6db07d7e3d9b2449a) the log at boot (you can see line 217 that they have a problem with fb0 --> Error opening /dev/fb0: No such file or directory). I have trying to resolve this but nothing to resolve... if you have an idea!

graugans, 2016-12-16:
My problem with Seco SBC A62 is there are the DRAM Settings missing in any official source. The one I do use may not optimal. But this shouldn't conflict with LVDs. Have you checked the manuals if the Solo has all capabilities? In UDOO Neo they use a special chip for HDMI AFAIK... So maybe the Solo is special as well.

modjo756, 2016-12-17:
Yes Graugans, if i boot on emmc (that have the seco original image), this work fine on lvds.

graugans, 2016-12-17:
Have you checked the device tree? Maybe there is something different?

modjo756, 2016-12-17:
Yes but nothing that can help for the moment ...

modjo756, 2016-12-17:
In this file there are some config for RAM https://github.com/UDOOboard/uboot-imx/blob/3989280f796a19ef4d0b6aa4a43f9ac53582a72b/board/seco/ddr/mx6_SBC_A62_ram.cfg