graugans/meta-udoo #34: CAN not working NEO

modjo756, 2016-12-17:
Hi with the new version of yocto (since u-boot fslc) i have some trouble with can bus. With the older version if i did this (with two shell connected to the card) : 
first shell : 

_root@udooneo:~# sudo ip link set can0 type can bitrate 125000 loopback on
root@udooneo:~# sudo ifconfig can0 up
root@udooneo:~# candump can0_

on the second shell : 

_root@udooneo:~# cansend can0 500#1E.00.10
root@udooneo:~# cansend can0 450#1E.00.20_

i obtain this on the first : 
  _can0  500   [3]  1E 00 10
  can0  450   [3]  1E 00 20_

all work fine !

Now try to reproduce the same with the new image (uboot fslc) : 

First shell : 
_root@udooneo:~# sudo ip link set can0 type can bitrate 125000 loopback on
root@udooneo:~# sudo ifconfig can0 up
root@udooneo:~# candump can0
interface = can0, family = 29, type = 3, proto = 1_

second shell : 
_root@udooneo:~# cansend can0 500#1E.00.10
interface = can0, family = 29, type = 3, proto = 1
root@udooneo:~# cansend can0 450#1E.00.20
interface = can0, family = 29, type = 3, proto = 1_

i obtain this result : 
_<0x001> [1] f4 
<0x001> [1] c2_ 

Now we can see a new line interface = can0, family = 29, type = 3, proto = 1
I didn't understand what's happen ? This is due to u-boot fslc ? ...

modjo756, 2016-12-17:
after another test, if i send the message like this : 
`root@udooneo:~# cansend can0 -i 0x500 0x10 0x20 0x30`
i obtain this : 
<0x500> [3] 10 20 30 

It seems that the version of canutils or libsocketcan has changed ... strange

modjo756, 2016-12-19:
Tested today with oscilloscope and peakcan (on another PC). We have a BUSHEAVY error in PCAN-view. I have a signal with oscilloscope but is not good, perhaps a problem with the clock ... @fmntf , just a question, did you make some special modifications in your u-boot-imx for neo flexcan please ?

fmntf, 2016-12-19:
Sorry, no CAN support in U-Boot.

modjo756, 2016-12-19:
Thanks for your answer !

cyrilfr, 2017-01-31:
Did you figure out how to fully support CAN?

modjo756, 2017-01-31:
No for the moment CAN not working ...

graugans, 2017-01-31:
@cyrilfr your work on kernel 4.1 looks really promissing. When I find some time I'll try to integrate your changes into my layer. At the moment I have some difficulties to test CAN due to a missing endpoint.

modjo756, 2017-04-19:
Just build an image (minimal/command line) with morty branch and your layer @graugans and CAN is not working, i have this error : 
```

root@udooneo:~# ip link set can0 type can bitrate 125000 loopback on
root@udooneo:~# ifconfig can0 up
root@udooneo:~# candump can0
interface = can0, family = 29, type = 3, proto = 1
socket: Address family not supported by protocol
```
@cyrilfr , did you try with can with morty Branch ?

For information i use kernel 3.14 and disable M4, with ifconfig -a i have this : 

`root@udooneo:~# ifconfig -a     
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP RUNNING NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:142 

can1      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:143 
`

modjo756, 2017-04-20:
Some progress, we need to use can-utils and not canutils (see in /images/udoo-image-full-cmdline.bb), now when i launch candump (in kernel 3.14, i suppose this works with kernel 4.1 i will test it tomorrow) i have this : 
`socket: Address family not supported by protocol`
and in dmesg : 
[  208.250295] flexcan 2090000.can can0: writing ctrl=0x0e313005
[  218.374684] can: no symbol version for sock_register
[  218.379687] can: Unknown symbol sock_register (err -22)

modjo756, 2017-04-21:
Ok, today i tried with kernel 4.1 and it work if i disable M4 core ! (perhaps a problem like describe [here](http://www.udoo.org/forum/threads/canopen-using-flexcan.6438/)). Now retry with kernel 3.14.56, disable m4 core but already the same error announced in my precedent post ... Help are welcome !

graugans, 2017-04-22:
@modjo756 do you have a kernel 4.1 recipe around? Can you provide a link?

modjo756, 2017-04-24:
@graugans , see [here](https://github.com/modjo756/kernel-udoo-4.1) (just created one)

modjo756, 2017-04-24:
Today, i'm back to krogoth branch and last revision of kernel 3.14.56 and CAN works if i disable or remove scketch from M4 (i don't know why on Morty branch this didn't work on kernel 3.14) ... There is probably a conflict with default arduino spi pins and CAN pins on A9 like describe on [neoDocs](http://www.udoo.org/docs-neo/Hardware_&_Accessories/SPI_bus.html) : 

`ECSPI 5 has the same pin order, function, and position as Arduino UNO. By default it is assigned to the Arduino 101-compatible M4 core.`

I have open an issue on [gitHub](https://github.com/UDOOboard/arduino-board-package/issues/5) for this. I hope @ektor5 or @UDOOboard can have a look on this ...