graugans/meta-udoo #38: wlan not working

modjo756, 2017-07-03:
Hi i'm trying to use wifi (host ap with connman) on neo card with morty branch and kernel 3.14.56 but i have this error when i'm trying to mount the network 

`root@udooneo:~#  connmanctl tether wifi on wifiNetwork 123456789
Wifi SSID set
Wifi passphrase set
[   88.217414] ipv6: no symbol version for sock_register
[   88.224577] ipv6: Unknown symbol sock_register (err -22)
[   88.231826] ipv6: no symbol version for ns_capable
[   88.236633] ipv6: Unknown symbol ns_capable (err -22)
Error enabling wifi tethering: Not supported`

it's strange because i use exactly the same defconfig (for kernel build) that i use on krogoth and on this branch all work fine ?! Do you have an idea where is the issue ?

cyrilfr, 2017-09-14:
There is a bug in the wl18 (udoo neo) driver I guess. When you try to connect multiple UDOO to the same AP, they get disconnected after a while. There is messages from the kernel.

wlcore: WARNING corrupted packet in RX: status: 0x3 len: 76 (repeated 2 times)
wlcore: WARNING corrupted packet in RX: status: 0x3 len: 216 (repeated 2 times)
wlcore: WARNING corrupted packet in RX: status: 0x3 len: 104 
wlcore: WARNING corrupted packet in RX: status: 0x3 len: 64
wlcore: WARNING corrupted packet in RX: status: 0x3 len: 64 
...

I updated the wl18-conf recipe to the last revision without success. I use the v 4.1.15 of the kernel.
I'll have a look on this tomorrow.

cyrilfr, 2017-09-21:
What's the point of the file "wl18xx-conf.bin" in the recipe wl18xx-conf ?

graugans, 2017-09-25:
The origin of ``wl18xx-conf.bin`` was the official UDOO image. @cyrilfr if you have a better version I would really happy to add this file. As far as I remember this file could be generated with the input of some ini files. I am not sure if it is possible to generate an ini back from the bin to compare what are the differences.

cyrilfr, 2017-09-25:
I don't have a better configuration file. As far as I know, this is more a firmware related issue, which is closed-source. There is a very ugly workaround proposed in this thread using a watchdog to detect corrupted packet and perform a reauthentication with WPA_supplicant : https://e2e.ti.com/support/wireless_connectivity/wilink_wifi_bluetooth/f/307/t/401452

I'm surprise that this WL18xx chipset simply doesn't fully work for years!

graugans, 2017-09-25:
That are bad news, I guess you've tried this with the official UDOO image as-well.

cyrilfr, 2017-09-26:
I was wrong, the firmware provided in poky isn't the last one. Poky provides the v8.9.0.0.69 and the last one available is the v8.9.0.0.75. This update fixed the issue.

cyrilfr, 2017-09-26:
You can find a fully working layer with linux v4.1 on my repo : https://github.com/cyrilfr/meta-udoo
WARNING: I updated a few options in the defconfig to fix issues from the graugans's layer and moved recipes to more relevant places.

cyrilfr, 2017-11-21:
There is still random errors with the last firmware (8.9.0.0.76) :
```
[ 1832.622482] wlcore: ERROR SW watchdog interrupt received! starting recovery.
[ 1832.629728] ------------[ cut here ]------------
[ 1832.634759] WARNING: CPU: 0 PID: 138 at /usr/src/debug/ti-compat-wireless-wl18xx/R8.7_SP2+gitAUTOINC+5c94cc59ba-r2/compat-wireless/drivers/net/wireless/ti/wlcore/main.c:806 wl12xx_queue_recovery_work.part.9+0x58/0x5c [wlcore]()
[ 1832.655391] Modules linked in: bnep btwilink bluetooth ipv6 usb_f_acm u_serial g_serial libcomposite wl18xx(O) wlcore(O) mac80211(O) cfg80211(O) rfkill mxc_vadc mx6s_capture mxc_dcic fxos8700 fxas2100x wlcore_sdio(O) compat(O) ci_hdrc_imx usbmisc_imx ci_hdrc udc_core ehci_hcd evbug galcore(O) st_drv
[ 1832.683567] CPU: 0 PID: 138 Comm: irq/77-wl18xx Tainted: G           O    4.1.15_2.0.x-udoo+g7773e46 #1
[ 1832.692964] Hardware name: Freescale i.MX6 SoloX (Device Tree)
[ 1832.698852] [<80015adc>] (unwind_backtrace) from [<80012600>] (show_stack+0x10/0x14)
[ 1832.706632] [<80012600>] (show_stack) from [<80513c10>] (dump_stack+0x74/0xb4)
[ 1832.713893] [<80513c10>] (dump_stack) from [<8002fd1c>] (warn_slowpath_common+0x80/0xb0)
[ 1832.721993] [<8002fd1c>] (warn_slowpath_common) from [<8002fde8>] (warn_slowpath_null+0x1c/0x24)
[ 1832.730849] [<8002fde8>] (warn_slowpath_null) from [<7f2067d8>] (wl12xx_queue_recovery_work.part.9+0x58/0x5c [wlcore])
[ 1832.741652] [<7f2067d8>] (wl12xx_queue_recovery_work.part.9 [wlcore]) from [<7f206a64>] (wlcore_irq+0x168/0x16c [wlcore])
[ 1832.752672] [<7f206a64>] (wlcore_irq [wlcore]) from [<80068f7c>] (irq_thread_fn+0x1c/0x40)
[ 1832.760972] [<80068f7c>] (irq_thread_fn) from [<80069280>] (irq_thread+0x12c/0x178)
[ 1832.768657] [<80069280>] (irq_thread) from [<800489ac>] (kthread+0xdc/0xf4)
[ 1832.775665] [<800489ac>] (kthread) from [<8000f3e8>] (ret_from_fork+0x14/0x2c)
[ 1832.782890] ---[ end trace cf641541ff17fd20 ]---
```
But it seems to recover after that.

patrykk, 2017-11-24:
I could not reproduce a problem:

> "root@udooneo01:~# dmesg  | grep -i wl
> [    1.004249] wlan-en-regulator: disabling
> [    7.964113] wlcore: wl18xx HW: 183x or 180x, PG 2.2 (ROM 0x11)
> [    7.997218] wlcore: loaded
> [    7.997260] wlcore: driver version: R8.6_SP1
> [   11.072756] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [   11.619405] wlcore: PHY firmware version: Rev 8.2.0.0.232
> [   11.682523] wlcore: firmware booted (Rev 8.9.0.0.48)
> [   11.705156] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [   12.493367] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
> [   13.865615] wlan0: authenticate with a0:21:b7:ac:8f:b9
> [   13.869838] wlan0: send auth to a0:21:b7:ac:8f:b9 (try 1/3)
> [   13.888457] wlan0: authenticated
> [   13.899149] wlan0: associate with a0:21:b7:ac:8f:b9 (try 1/3)
> [   13.903066] wlan0: RX AssocResp from a0:21:b7:ac:8f:b9 (capab=0x431 status=0 aid=1)
> [   13.912543] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
> [   13.917221] wlan0: associated
> [   14.008846] wlcore: Association completed.
> "

cyrilfr, 2017-11-27:
It appears randomly.

graugans, 2017-11-29:
That's strange. Sadly I can't take a look at this before XMas holidays

cyrilfr, 2018-05-22:
I fxed this issue in the rocko branch: https://www.udoo.org/forum/threads/meta-udoo-rocko-for-udoo-neo.14671/