graugans/meta-udoo #39: upstream/mainline u-boot/kernel

RobertBerger, 2017-07-05:
Hi,

I am looking into the udoo neo in order to play with LInux on the i.mx6 + FreeRTOS on the Cortex M4.

... but i would like to use upstream/mainline u-boot and kernel.

As far as I can see there is udoo support in u-boot[1], but not sure this works for the 

[1] http://git.denx.de/?p=u-boot.git;a=blob;f=board/udoo/README;h=6fbcc598f77b2fa19689a7fc6f84548b08f6348f;hb=HEAD

graugans, 2017-07-05:
Hello @RobertBerger this is a topic I have on the TODO List since a long time. It would be a real plus to get mainline u-boot integrated into  this layer. I am highly interested in FreeRTOS on the M4 as-well. Some greetings from **ifm**

modjo756, 2017-07-05:
Like you it would be great to have a mainline kernel and FreeRTOS on M4 but i cry that the big problem is the driver for rpmsg between A9 and M4 ...

RobertBerger, 2017-07-09:
@modjo756: I don't think this is the problem. There are reference implementations for the i.mx7. I assume the part running on the Cortex M4 should be pretty similar. 
Have a look here:
https://github.com/OpenAMP/open-amp/wiki/AMP-Intro
https://stash.phytec.com/projects/PUB/repos/freertos-phytec/commits?until=refs%2Fheads%2Fv8.0.0-phy&merges=include
http://git.toradex.com/cgit/freertos-toradex.git/tree/
The Linux drivers should be pretty much an out of the box experience, the Cortex M4 stuff might need some adjustment.

modjo756, 2017-08-30:
@RobertBerger , for information there is an update to use kernel 4.1 with M4 and rpmsg [here](https://github.com/graugans/meta-udoo/issues/18#issuecomment-325576153) . Otherwise did you make some progress on your side ?

RobertBerger, 2017-08-31:
@modjo756 , I did not have any time to play with this yet. At the moment I am at a board manufacturer. They have an i.mx7/Cortex-M4 board. This might be an option as well.

sarnold, 2017-11-12:
I like pushing mainline/FOSS onto these things, so I made some test builds, one by hand and one using meta-udoo plus my own layers, all FOSS with no vendor trees (just machine info).  Mainline u-boot has a minor config/env patch and mainline kernel (4.12.14) has some initial imx_rproc patches.  Everything works pretty well so far, but most of my testing so far has been kernel/u-boot config and getting the proper foss config and/or override bits into the manifest build (meaning it would be good if someone could test the rproc stuff, etc).

The manual build steps are mainly stolen from Robert's Udoo page; the gist is here:

https://gist.github.com/sarnold/2e244fa8580ec715321a515c72535d4f

and fork of the build repo with imx_rproc patches:

https://github.com/VCTLabs/armv7-multiplatform

I used debian stretch since Robert maintains the kernels and a bunch of useful embedded tools in his arm repo, but ubuntu should also work.  I also tested the comitup wifi setup, which also "works" well enough.

The manifest build uses the same kernel/-uboot as above and is pretty much a "normal" poky/oe tree, so it's not like the usual fsl-based manifest build.  In this case the top level dir is poky/ with all the metadata layers under that.  Just use the normal oe-setenv script.  There are also a couple of small changes to meta-udoo to allow some overrides; my working build configs are in the conf/ dir in the meta-small-arm-extra layer.  The manifest only has the morty branch so far:

https://github.com/sarnold/foss-udoo-neo-platform-bsp

and uses the morty-4.1 branch in meta-udoo:

https://github.com/sarnold/meta-udoo

Let me know if you want a PR for the meta-udoo changes (but I guess you might want to look things over first ;)

sarnold, 2017-11-12:
Also note the device tree support in mainline u-boot/kernel is only the minimum 3 board types without any of the extra subtypes in the fsl/udoo forks.  The rproc support should be enough to load firmware and start the M4 core (at least according to the discussion on the kernel list).  My rproc experience is all BeagleBone so far but I do have some arduino/M4 experience, so as long as the linker script is correct it should work In Theory...

RobertBerger, 2017-11-22:
@sarnold How about moving to 4.14? Looks like the rproc patch made it upstream[1]

[1] https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/tree/drivers/remoteproc/imx_rproc.c?h=v4.14.1

cyrilfr, 2017-11-29:
Buildroot uses the v4.10 with the imx6_v6_v7 configuration file.

[image]

modjo756, 2017-12-08:
It seems they have some progress on mainline u-boot/kernel, good !!! 
A question, i have tested kernel 4.1.15 (from francesco) but rpsmg driver not working for the moment, did someone have a working driver on this point ? Otherwise, what is rproc ?

sarnold, 2017-12-11:
I've been testing some newer recipes (mainly on udoo quad) but got hung up on a couple of things, one is that newer-than-4.12 kernels actually don't boot fully with my "typical" rcn-style uEnv args.  Turns out they need the extra (atypical) "uart_from_osc clk_ignore_unused" args to boot.  Also the udooqdl dts somehow lost the second uart definition for the internal sam3 interface so I added a patch for that.  Now that it looks pretty good, I need to bump the 4.14 recipe and throw the kernel/u-boot stuff in meta-udoo and push a branch.  Although I'll probably work in Arduino IDE first...

sarnold, 2017-12-14:
And I spent too much time hacking ardunio ide 1.8.5 with udoo/sam hardware tree and it sort of works with crossdev multilib build of arm-none-eabi-gcc 6.4.0.  Except for uploading the sketch, is there a generic mechanism for that with 4.14 kernel?  There is no /dev/udoo_whatever any more; do I need to port another patch?  Do I need I need to physically patch with jumper wires?  It's been a long time since I played with sketches on this thing (meaning udoo quad) so I could use a few tips...

sarnold, 2017-12-14:
Also I made a wiki page for this:

https://wiki.gentoo.org/wiki/Udoo

sarnold, 2017-12-15:
Got it, found the right forum post and made a driver/dts patch and now arduino IDE on Gentoo with mainline can upload a sketch.  Now I just need to find an LED for blinky test...

sarnold, 2019-03-25:
The serial/ardiuno patches for imx6 udoo quad/dual are still in Robert's kernel builder repo: https://github.com/RobertCNelson/armv7-multiplatform and I can at least see the devices running 5.0.1 (sadly I have no time to test it with arduino stuff right now).  If someone followed the wiki page for Udoo here:  https://www.digikey.com/eewiki/display/linuxonarm/UDOO and used the current 5.0.1x version they could easily test upstream debian/ubuntu...