graugans/meta-udoo #40: Qtquick extras missing in morty branch with qt 5.8

modjo756, 2017-07-27:
I have build qt 5.8 in morty branch but it appears that qtquick-extras (plugin of qtquickcontrols) is not build ... I have included qtquickcontrols and qtquickcontrols-qmlplugins but the package is not build like in 5.7 ... Don't find solution for the moment !

graugans, 2017-07-28:
When there is a PR from @drewmoseley I will merge it. At the moment I do not have time to work on a pyro branch on my own. Maybe I have something floating around and check if I can push it upstream.

drewmoseley, 2017-07-28:
@graugans there is no pyro branch to submit a PR against.  I do have a working one in my fork.  I'll take a look later today and make sure it's still functional and post the link here.

drewmoseley, 2017-07-30:
Apologies for taking a bit longer on this than I had anticipated.  I was finally able to complete my builds and test these branches today.  

My wip-pyro branch seems to work based on the minimal testing I've done.

I did see some issues with my wip-master branch; notably an RCU stall.  Also there is a build issue due to a change in poky that requires an additional change in meta-freescale.  My hackish workaround for that is here: https://github.com/drewmoseley/meta-freescale/commit/dafceb34df088b597e3fb72ef1e14d0d66e3d142

Here are my meta-udoo WIP branches:
- https://github.com/drewmoseley/meta-udoo/tree/wip-pyro
- https://github.com/drewmoseley/meta-udoo/tree/wip-master

@graugans how should I submit these since there is no existing branch for a PR?  They are both straight on top of the existing morty branch without too many complications.

drewmoseley, 2017-08-08:
@graugans how should I submit the above branches for consideration?  I don't want to do a PR against morty for obvious reasons.

graugans, 2017-08-10:
Sorry for the delay, I'll create one

graugans, 2017-08-10:
I've just created the branch and the PR #41 and already merged it.

modjo756, 2017-08-10:
Back from holidays :-(, i will test and let you known !

modjo756, 2017-08-11:
@graugans could you create pyro branch for fsl-community-bsp-platform and udoo-community-bsp-base please ?

graugans, 2017-08-14:
@modjo756 done

modjo756, 2017-08-14:
@graugans @drewmoseley  I have launch a build in pyro branch but i have some issues during the generation of rootfs, see [here](https://gist.github.com/modjo756/a4afdb003e7acc3c7f6ae433c65af791).  This is the [local.conf](https://gist.github.com/modjo756/db352a7325ced3d581fde872dddbcf7a) that i use. The alsa-lib is present now in folder
> cortexa9hf-neon-mx6qdl-poky-linux-gnueabi

in krogoth/morty the alsa lib is in 
> udooqdl-poky-linux-gnueabi

Otherwise, in my recipe [enocean](https://github.com/modjo756/meta-iot/blob/master/recipes-enocean/enocean/enocean_1.9.0.bb) that work also on krogoth/morty now the libEOLink.la (present on build directory) is not copied in packages-split/enocean see under : 

├── enocean
├── enocean-dev
│   └── usr
│       └── include
│           ├── enocean
│           │   ├── api
│           │   │   ├── sec_api.h
│           │   │   └── sec.h
│           │   ├── eoAbstractMessage.h
│           │   ├── eoApiDef.h
│           │   ├── eoArchive.h
│           │   ├── eoArchiveTXT.h
│           │   ├── eoChainedMessage.h
│           │   ├── eoChainedReManMessage.h
│           │   ├── eoChannelEnums.h
│           │   ├── eoChannelInfo.h
│           │   ├── eoChunkMessage.h
│           │   ├── eoConverter.h
│           │   ├── eodBmFilter.h
│           │   ├── eoDebug.h
│           │   ├── eoDevice.h
│           │   ├── eoDeviceManager.h
│           │   ├── eoEEProfile.h
│           │   ├── eoERP2Converter.h
│           │   ├── eoFilterFactory.h
│           │   ├── eoGateway.h
│           │   ├── eoGenericProfile.h
│           │   ├── eoGpChainer.h
│           │   ├── eoGPChannelInfo.h
│           │   ├── eoGPHelper.h
│           │   ├── eoHalTypes.h
│           │   ├── eoHeader.h
│           │   ├── eoIDFilter.h
│           │   ├── eoIFilter.h
│           │   ├── eoISerialize.h
│           │   ├── eoISimpleBidirectionalProfile.h
│           │   ├── eoIWatcher.h
│           │   ├── eoLinuxPacketStream.h
│           │   ├── eoLogger.h                                                                                                                                                             
│           │   ├── eoManufacturer.h                                                                                                                                                       
│           │   ├── eoMessage.h                                                                                                                                                            
│           │   ├── eoMessageHandler.h                                                                                                                                                     
│           │   ├── eoMSCHelper.h                                                                                                                                                          
│           │   ├── eoPacket.h                                                                                                                                                             
│           │   ├── eoPacketStream.h                                                                                                                                                       
│           │   ├── eoProc.h                                                                                                                                                               
│           │   ├── eoProfileFactory.h                                                                                                                                                     
│           │   ├── eoProfile.h                                                                                                                                                            
│           │   ├── eoReCom.h                                                                                                                                                              
│           │   ├── eoReManChainer.h                                                                                                                                                       
│           │   ├── eoReManMessage.h                                                                                                                                                       
│           │   ├── eoRemoteManager.h                                                                                                                                                      
│           │   ├── eoRollingCodeStorage.h                                                                                                                                                 
│           │   ├── eoSecTeachInChainer.h
│           │   ├── eoSecureInfo.h
│           │   ├── eoSecurity.h
│           │   ├── eoSerialCommand.h
│           │   ├── eoStorageManager.h
│           │   ├── eoTeachInModule.h
│           │   ├── eoTelegramERP2.h
│           │   ├── eoTelegram.h
│           │   ├── eoTimer.h
[... 33 lines cut ...]
│           │       ├── eoEEP_D220xx.h
│           │       ├── eoEEP_D230xx.h
│           │       ├── eoEEP_D231xx.h
│           │       ├── eoEEP_D232xx.h
│           │       ├── eoEEP_D233xx.h
│           │       ├── eoEEP_D240xx.h
│           │       ├── eoEEP_D250xx.h
│           │       ├── eoEEP_D2A0xx.h
│           │       ├── eoEEP_D2B0xx.h
[... cut ...] eoEEP_D500xx.h
│           │       ├── eoEEP_F601xx.h
│           │       ├── eoEEP_F602xx.h
│           │       ├── eoEEP_F603xx.h
│           │       ├── eoEEP_F604xx.h
│           │       ├── eoEEP_F605xx.h
│           │       ├── eoEEP_F610xx.h
│           │       ├── eoF6EEProfile.h
│           │       └── eoSpecialProfile.h
│           └── eoLink.h
└── enocean-staticdev
    └── usr
        └── lib
            └── libEOLink.a

If you can have a look in you see an error on my recipe please ?

drewmoseley, 2017-08-14:
@modjo756 is there a reason you are trying to call the autotools stuff directly rather than just inheriting the autotools bbclass and letting it do everything?  See for instance https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/console-tools/console-tools_0.3.2.bb

modjo756, 2017-08-14:
@drewmoseley   thanks for your answer but can you explain more about that that please ? i suppose you are talking about my enocean.bb ?

drewmoseley, 2017-08-14:
Yes.  You should be able to simply inherit autotools and let that do all the recipe steps for you. That should set up the dependencies on autoconf-native, etc and generate the appropriate do_configure, do_compile, etc steps without you needing to do much of anything.

This may give you an idea: https://www.yoctoproject.org/docs/2.3.1/dev-manual/dev-manual.html#new-recipe-autotooled-package

modjo756, 2017-08-14:
Thanks a lot ! for enocean.bb this work fine but i already have the error for alsa-lib : 
ERROR: udoo-image-qt5-1.0-r0 do_rootfs: Could not invoke dnf. Command '/media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/recipe-sysroot-native/usr/bin/dnf -y -c /media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/rootfs/etc/yum.repos.d --repofrompath=oe-repo,/media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/oe-rootfs-repo --installroot=/media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/rootfs --setopt=logdir=/media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/temp --nogpgcheck install alsa-lib alsa-state alsa-tools apache2 base-passwd libcairo2 cam-user cinematicexperience cpufrequtils dbus-1 dnf libeolink0 expect libfontconfig1 libfreetype6 fsl-alsa-plugins gdbserver htop i2c-tools iftop minicom nano ntp ntp-utils ntpdate openssh-sftp-server openssl openzwave packagegroup-base packagegroup-base-usbhost packagegroup-core-boot packagegroup-core-full-cmdline packagegroup-core-sdk packagegroup-core-ssh-openssh packagegroup-fsl-gstreamer1.0-full packagegroup-fsl-tools-benchmark packagegroup-fsl-tools-gpu packagegroup-fsl-tools-testapps pango php psplash pulseaudio qt3d qt3d-qmlplugins qt5-env qtbase-plugins qtbase-tools qtcanvas3d libqt5charts5 qtconnectivity-qmlplugins qtdeclarative qtdeclarative-qmlplugins qtdeclarative-tools qtgraphicaleffects qtgraphicaleffects-qmlplugins qtimageformats-plugins qtlocation-plugins qtlocation-qmlplugins qtmultimedia qtmultimedia-plugins qtmultimedia-qmlplugins libqt5declarative5 libqt5declarative-plugins libqt5declarative-qmlplugins qtquickcontrols qtquickcontrols-qmlplugins qtquickcontrols2 qtscript libqt5sensors5 libqt5serialbus5 libqt5serialport5 libqt5svg5 libqt5svg-plugins qtsystems qtsystems-qmlplugins qtsystems-tools qttools-plugins qtvirtualkeyboard qtwebkit qtwebkit-examples-examples qtwebkit-qmlplugins libqt5xmlpatterns5 resize-rootfs rpm run-postinsts samba shadow sntp tree ttf-dejavu-common ttf-dejavu-sans ttf-dejavu-sans-condensed ttf-dejavu-sans-mono ttf-dejavu-serif ttf-dejavu-serif-condensed udev vsftpd locale-base-en-us locale-base-en-gb' returned 1:
Added oe-repo repo from file:///media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/oe-rootfs-repo
Last metadata expiration check: 0:00:01 ago on Mon Aug 14 14:24:40 2017 UTC.
No package alsa-lib available.
Error: Unable to find a match

ERROR: udoo-image-qt5-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /media/modjo/data1TO/yocto/pyro/quadBuild/tmp/work/udooqdl-poky-linux-gnueabi/udoo-image-qt5/1.0-r0/temp/log.do_rootfs.19900
ERROR: Task (/media/modjo/data1TO/yocto/pyro/sources/meta-udoo/qt5-layer/recipes-qt/images/udoo-image-qt5.bb:do_rootfs) failed with exit code '1'

modjo756, 2017-08-14:
After check, enocean now build .so lib and not .a and .la files : 

└── usr
    └── lib
        ├── libEOLink.so -> libEOLink.so.0.0.0
        ├── libEOLink.so.0 -> libEOLink.so.0.0.0
        └── libEOLink.so.0.0.0

I need to check on the card if it works !

graugans, 2017-08-14:
@modjo756 is your updated enocean recipe somewhere available? Can please verify is alsa is in distro features?

```
DISTRO_FEATURES_append = "alsa"
```

modjo756, 2017-08-14:
@graugans this is my enocean recipe (not commit for the moment) : 

SUMMARY ="enOcean recipe for yocto"

DESCRIPTION = "enOcean allow to control device that use this protocole for home automation ..."

AUTHOR = "modjo.buggy22@orange.fr"

LICENSE_FLAGS = "trial_enocean"

LICENSE = "CLOSED"

LIC_FILES_CHKSUM = " \
    file://${WORKDIR}/EOLink/LICENSE;md5=bad3069f1239057481be546550adfeca \
    file://${WORKDIR}/EOLink/LICENSE_TRIAL;md5=f812f2a3f8d3e12164316914e903e622 \
"

SRC_URI = "file://EnOceanLinkTrial_1_9_0_0.zip \
          file://0001-change-default-var-pi-script.patch \
          file://0002-change-path-includes-files.patch \
"

S = "${WORKDIR}/EOLink"

inherit autotools gettext

I added alsa to distro_features... waiting the build finished

modjo756, 2017-08-14:
Same result when i add alsa to distro_feature ...

No package alsa-lib available.
Error: Unable to find a match

drewmoseley, 2017-08-14:
@modjo756 what happens if you just try to "bitbake alsa-lib"?  My guess is that it has something to do with the LICENSE_blah settings.

graugans, 2017-08-16:
Another possibility is that including another alsa package contains the lib.

modjo756, 2017-08-16:
@drewmoseley , sorry for my answer delay ... 
test with "bitbake alsa-lib" --> build without error or warning ! but same error during rootfs !

modjo756, 2017-08-16:
@graugans , perhaps this [recipe](https://layers.openembedded.org/layerindex/recipe/65158/) from freescale can resolve this issue

modjo756, 2017-08-16:
i (think) i have resolved the problem with alsa like this. I created a udoo-image-qt5.bbappend on my layer : 

QT_TOOLS_remove = " qtbase-fonts\
                    qtdeclarative-plugins \
"
QT_TOOLS_append = " \
    qtquick1 \
    qtquick1-qmlplugins \
    qtquick1-plugins \
    qtserialport \
    qtserialbus \
    qtcharts \
    qtquickcontrols \
    qtquickcontrols-qmlplugins \
    qtquickcontrols2 \
    qtvirtualkeyboard \
    qttools-plugins \
    qtxmlpatterns \
    qtcanvas3d \
    qtgraphicaleffects \
"

IMAGE_INSTALL_remove = "alsa-lib fsl-alsa-plugins"
IMAGE_INSTALL_append = "imx-alsa-plugins"

Graugans in meta-udoo/qt5-layer you can remove the folder qt5 that contain a recipe for serialbus (not needed with the last version of qt and it cause an error during the build).
About u-boot in file meta-udoo/recipes-bsp/u-boot/u-boot-fslc/0001-UDOO-Quad-Dual-support.patch i have changed this : 
line 1462 +	#define CONFIG_MMCROOT	"/dev/mmcblk0p2" by this
line 1462 +	#define CONFIG_MMCROOT	"/dev/mmcblk0p1" 

line 1480+	"mmcpart=2\0" \ by this 
line 1480+	"mmcpart=1\0" 

without this change the card won't boot !

@graugans  @drewmoseley The [log](https://gist.github.com/modjo756/f803161e629bcee293af9aca4d845c39) of the first boot

Yet a problem with uEnv.txt the file is not read because it load the default settings video_output=hdmi and on my config i load lvds15 ...

modjo756, 2017-08-23:
I close this issue now qt5.8 work fine !

closed 2017-08-23 by modjo756