graugans/meta-udoo #42: .wic image for yocto

sagniknitr, 2017-11-09:
Is there any specific reason for using the .wic image for yocto in udoo boards , and not .sdcard image ?

graugans, 2017-11-09:
The ``.wic`` utility is build into YOCTO the ``.sdcard`` stuff is a class introduced by the raspberry pi meta-layer as far as I know. For me it was easier to use the already available wic mechanism. After getting rid of the vfat partition for the u-Boot script there was no real usecase for the  ``.sdcard`` hack anymore.

Or did I miss understood the question?

sarnold, 2017-11-12:
Actually this came up right before I started hacking on your layer so it was a good chance for me to test it.  Anyway, apparently Otavio said something about wic replacing the bbclass upstream as the new/default method.

graugans, 2017-11-13:
@sarnold and @sagniknitr so we can close this?

sarnold, 2017-11-15:
Fine by me, did you want a pull request for mainline kernel/u-boot stuff?

graugans, 2017-11-16:
That would be really nice

sarnold, 2017-11-17:
I should probably add a parameter for using mainline, since your image recipes try to install the fsl graphics demos (even the command line image).  I just used(abused) core-image-minimal with most of the config/image_install in local.conf (samples are in meta-small-arm-extra conf/ dir).

Starting a separate issue for mainline integration seems like a good idea, since for now it's all in the manifest repo and meta-small-arm-extra (https://github.com/sarnold/meta-small-arm-extra).  How does that sound?

closed 2017-11-19 by sagniknitr

graugans, 2017-11-21:
That sounds perfect.

sarnold, 2017-12-07:
I added latest kernel recipes (probably a teensy bit behind now) and tested on neo/udoo-quad, and added that hack parameter for mainline so I could build your images.  I did notice freescale has a use_mainline thing but I didn't look at it (or test it) yet, mainly since my funky small-arm layer already builds the free armada/etnaviv graphics stuff by masking out some freescale recipes and such.  Also I started adding the mainline recipes to my meta-udoo fork but then I got sidetracked; you should have a PR to look at sometime soon...