graugans/meta-udoo #45: qt5 can't display with distro fslc-x11

embeddedvn, 2018-03-19:
Hello,

I am trying to use distro fslc-x11 with udoo-image-qt5 but it can't display, I don't know exactly the reason why? 
root@udooneo:~# export DISPLAY=:0
root@udooneo:~# rxhost +
xhost:  unable to open display ":0"
root@udooneo:~# Qt5_CinematicExperience                       
QXcbConnection: Could not connect to display :0               
/usr/bin/Qt5_CinematicExperience: line 4:   488 Aborted

the same with x11vnc, it can't connect to display. 
with distro fslc-framebuffer, it can work well.

Thanks!

modjo756, 2018-03-20:
Hi embeddedvn, i never use x11 with yocto, sorry ... Perhaps you can try with wayland and have a look on bootToQt layer for this !

graugans, 2018-03-20:
@embeddedvn X11 was never a use-case why is x11 needed?

embeddedvn, 2018-03-20:
Hi graugans, modjo,

I want to view Qt app by remote desktop like x11vnc and view on LCD hdmi also. Do you know any way to remote like that by using another distro? with framebuffer distro I can run Qt graphics on hdmi lcd but can't remote vnc. I tried to build framebuffer-vncserver recipe but I met some error when build it. 

Thanks for your respond
Dong

modjo756, 2018-03-20:
A solution will be (depending your application complexity) to realize your app with qtwebengine (web server). Like this you can see your web site with web engine on the card and use a web browser on another PC to access your app ...

modjo756, 2018-03-20:
To demonstrate an example of my recent post see [here](http://www.phytec.de/documents/html5-kit-quickstart/#The_Rapid_Development_Kit_content). I have tested this image on phytec board and it works fine. The web page appears on the embedded screen. If i connect to the board with a pc i have the same HMI !
Otherwise i think you must have a look on weston (if you have a raspberry you can test boot to Qt to test it)
http://doc.qt.io/QtForDeviceCreation/qtee-custom-embedded-linux-image.html

embeddedvn, 2018-03-22:
Thanks for your suggest, modjo. I will try that solution. But I really like to use x11vnc on udooneo like this way: https://www.udoo.org/docs-neo/Basic_Setup/Remote_Desktop_(VNC).html

graugans, 2018-03-22:
@embeddedvn you should definitly think about the approach from @modjo756 it is much more modern approach I guess the X11 server + VNC is not get the performance you might expect and you need a separate application to view it. Remember the Neo is single core CPU.