graugans/meta-udoo #46: UDOO quad can bus issues

Tal-eigen, 2018-03-27:
Hello,

I am currently having issues getting UDOO's flexcan working. 

Following is the yocto build configuration i am using. ( krogoth branch) 

Build Configuration:
BB_VERSION = "1.30.0"
BUILD_SYS = "x86_64-linux"
NATIVELSBSTRING = "universal"
TARGET_SYS = "arm-poky-linux-gnueabi"
MACHINE = "udooqdl"
DISTRO = "poky"
DISTRO_VERSION = "2.1.3"
TUNE_FEATURES = "arm armv7a vfp thumb neon callconvention-hard cortexa9" 
TARGET_FPU = "hard" 
meta 
meta-poky = "HEAD:444dc2e99b3c3967d9f83380c34bb99077a6ffa8" 
meta-oe 
meta-multimedia 
meta-python 
meta-networking = "HEAD:55c8a76da5dc099a7bc3838495c672140cedb78e" 
meta-ros = "master:e2566402ab108a19634354a934788109422cf409" 
meta-fsl-arm = "HEAD:e2254e7b2ded0c2b66b1226f879b3a6d52037b2d" 
meta-fsl-arm-extra = "HEAD:2c28e636ec15c2cfd49bc9cebe0bbbcfde95bc7b" 
meta-fsl-demos = "HEAD:a165068f8a0d1cf29aabe4b4053f28be1c2aa492"
meta-udoo = "HEAD:a01cfce52c52fc942eab4375cd249795ca24c53f"

I've used the following patch to enable the can bus

```
From 3d23716944adbe2495a786d0021288d8a9624e67 Mon Sep 17 00:00:00 2001
From: Mohammed talha <mohammedtalha89@gmail.com>
Date: Fri, 9 Mar 2018 01:11:52 +0530
Subject: [PATCH] arch/arm/boot/dtc/im6qdl-udoo-externalpins.dtsi - Added CAN
support

---
arch/arm/boot/dts/imx6qdl-udoo-externalpins.dtsi | 6 +++---
1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/arch/arm/boot/dts/imx6qdl-udoo-externalpins.dtsi b/arch/arm/boot/dts/imx6qdl-udoo-externalpins.dtsi
index b45b5eb..34b5a3b 100644
--- a/arch/arm/boot/dts/imx6qdl-udoo-externalpins.dtsi
+++ b/arch/arm/boot/dts/imx6qdl-udoo-externalpins.dtsi
@@ -66,8 +66,8 @@
MX6QDL_PAD_EIM_CS1__GPIO2_IO24 0x80000000 // {{external-gpio-51}}
MX6QDL_PAD_EIM_CS0__GPIO2_IO23 0x80000000 // {{external-gpio-52}}
MX6QDL_PAD_EIM_D24__GPIO3_IO24 0x80000000 // {{external-gpio-53}}
- MX6QDL_PAD_GPIO_8__GPIO1_IO08 0x80000000 // {{external-gpio-54}}
- MX6QDL_PAD_GPIO_7__GPIO1_IO07 0x80000000 // {{external-gpio-55}}
+ //MX6QDL_PAD_GPIO_8__GPIO1_IO08 0x80000000 // {{external-gpio-54}}
+ //MX6QDL_PAD_GPIO_7__GPIO1_IO07 0x80000000 // {{external-gpio-55}}
>;
};

@@ -161,7 +161,7 @@
&flexcan1 {
pinctrl-names = "default";
pinctrl-0 = <&pinctrl_flexcan1>;
- status = "disabled";
+ status = "okay";
};

&usdhc1 {
-- 
2.7.4
```

It seems to work. Since i get the following CAN related log in my dmesg output

: dmesg | grep 'can'
[ 1.156890] ahci-imx 2200000.sata: SSS flag set, parallel bus scan disabled
[ 7.759429] 2090000.can supply xceiver not found, using dummy regulator
[ 7.767589] flexcan 2090000.can: device registered (reg_base=c0ae8000, irq=142)
[ 85.230994] flexcan 2090000.can can0: writing ctrl=0x0e312005
[ 376.099239] can: controller area network core (rev 20120528 abi 9)
[ 376.120856] can: raw protocol (rev 20120528)
[ 1347.862670] flexcan 2090000.can can0: writing ctrl=0x0e313055
[ 1347.868676] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready

Line 2 of the dmesg output says supply transceiver not found. Is there something missing in my device tree configuration. I have tried can-utils. candump is able to echo back the msgs sent using cansend when loopback is enabled. But Scope probing shows no activity on the Tx and Rx pins. I have tried using sn65hvd230 can transceiver without any success. The same transceiver works fine on other stm32 devices. Please do let me know if there is something that i could try in order to get this working.

modjo756, 2018-03-27:
Hi i don't think that your Line2 is a problem. Can you copy here an ifconfig and ifconfig -a please ?

Tal-eigen, 2018-03-27:
Here is the output of ifconfig:
```
root@udooqdl:~# ifconfig 
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP RUNNING NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:142 

eth0      Link encap:Ethernet  HWaddr 00:c0:08:88:23:92  
          inet addr:192.168.1.131  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2c0:8ff:fe88:2392/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:798 errors:0 dropped:0 overruns:0 frame:0
          TX packets:492 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:77762 (75.9 KiB)  TX bytes:62141 (60.6 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:2 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:140 (140.0 B)  TX bytes:140 (140.0 B)
```
after executing "cansequence can0" ifconfig out changes to:
```
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP RUNNING NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:11 (11.0 B)
          Interrupt:142 
```
for the following command:
`cansend can0 5A1#11.22.33.44.55.66.77.88`

The expected candump output is:
`can0  5A1  [8] 11 22 33 44 55 66 77 88`

But all i end up getting is the following:
`<0x001> [1] 05 `.

After going through another link here 
[(https://community.nxp.com/thread/437574)]
It says that we need to add the following to the imx6qdl-udoo.dtsi file:

```
reg_can_xcvr: regulator@5 {
compatible = "regulator-fixed";
reg = <5>;
regulator-name = "CAN XCVR";
regulator-min-microvolt = <3300000>;
regulator-max-microvolt = <3300000>;
pinctrl-names = "default";
pinctrl-0 = <&pinctrl_flexcan_xcvr>;
gpio = <&gpio4 21 GPIO_ACTIVE_HIGH>;
enable-active-low;
};
```
Since the sabresd uses the same imx6q processor as udoo. Could this be the reason behind the xceiver not found error msg.

Regards

modjo756, 2018-03-27:
Ok. Did you make an (for example)

```ip link set can0 type can bitrate 125000 loopback on```
```ifconfig can0 up```
Before cansend ?

Can you test : 
```ip -s -d link show can0```

Tal-eigen, 2018-03-27:
Yes i did configure the interface. Following is the output of the requested command:
```
root@udooqdl:~# ip -s -d link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 
    can <LOOPBACK> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 0 
          bitrate 500000 sample-point 0.866 
          tq 133 prop-seg 6 phase-seg1 6 phase-seg2 2 sjw 1
          flexcan: tseg1 4..16 tseg2 2..8 sjw 1..4 brp 1..256 brp-inc 1
          clock 30000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          0          0          0         
    RX: bytes  packets  errors  dropped overrun mcast   
    0          0        0       0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    20         17       0       0       0       0  
```

modjo756, 2018-03-28:
It's strange, i didn't use CAN on udoo Quad but on neo and secoA62 and have the same result ... but my CAN network is working ! It seems that i already have a problem like you due to the use of canutils instead of can-utils in my yocto image ... Try to change that and let me know !

Tal-eigen, 2018-03-28:
I know . Everything seems to be in order. I let it run for a while to see if eventually any errors come up and here is the result.

```
root@udooqdl:~# ip -s -d link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 
    can <LOOPBACK> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 0 
          bitrate 125000 sample-point 0.875 
          tq 500 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
          flexcan: tseg1 4..16 tseg2 2..8 sjw 1..4 brp 1..256 brp-inc 1
          clock 30000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          0          0          0         
    RX: bytes  packets  errors  dropped overrun mcast   
    0          0        0       0       0       0       
    TX: bytes  packets  errors  dropped carrier collsns 
    31320      5504     0       0       0       0   
```    

no errors at all. dmesg doesnt throw up any errors either except for the 
`[    3.056323] 2090000.can supply xceiver not found, using dummy regulator
`
and i switched over to the can-utils package now. The issue persists. So this clearly isnt a software related issue. I wonder if anyone has got can bus working on udoo quad before. i do have an old Unico-kernel build. I'll try using that to rule out the possibility of a software issue. 

I've looked at the schematic (Revision D) and this has piqued my interest 

[image]
The sam3x8e cantx0 and canrx0 (pa0 and pa1) lines are directly connected to the imx6 flexcan tx and rx lines via a 33 ohm resistor. I dont know if this is the main culprit but i would like to think that the sam3x8e lines are causing the issue by being in a default tri-state high impedance mode. But before i hack the board i'd like to know if anyone has got the flexcan interface working on udoo quad before. 

Regards

Tal-eigen, 2018-03-29:
I did a little bit of hardware tinkering yesterday and and i was wrong regarding the r1029 and r1030 resistors. Following schematic shows the connection of CANRX_IMX and CANTX_IMX signals (which are also connected to the CANRX0 and CANTX0 signals of the sam3x8e chip) connected to the R3 and R5 balls of the IMX6Q referenced below.

[image]

But according to the IMX6Q reference manual (IMX6DQRM Rev. 4, 09/2017) page 1293. GPIO_7 is Flexcan1_Tx. 

Ofcourse i could be wrong and this could just be an error in the schematic. But i dont understand what else could be causing this issue.

graugans, 2018-03-29:
Soory I never tried CAN on any of the UDOO boards

Tal-eigen, 2018-03-29:
I just tried with the official udoo distribution UDOObuntu 2.1.4 and enable CAN using the device tree editor. Doesn't work. I guess i'm just going to remove the sam3x8e chip from the board, i never use it anyways. I'll keep you guys posted.

Tal-eigen, 2018-03-29:
I went ahead and de-soldered the sam3x8e chip to see if it helps. 
[image]

It doesnt :( . I'm out of ideas now. Please do let me know if you guys have something that i could try. In the meanwhile i'll be getting an udoo neo soon. @modjo756 has gotten can working on it i suppose. 

Regards.

Tal-eigen, 2018-03-31:
@modjo756  I've looked at the link you provided and it is for a pre 3.1x kernel i suppose. Since now all the HW modifications are done using dts files. Regarding udoo neo are you using 4.4 kernel or 3.14? and do both the can interfaces work out of the box with the current morty branch?

closed 2018-04-02 by Tal-eigen

modjo756, 2018-04-03:
@Tal-eigen , did you solved this issue ?
Concerning neo, i use kernel 3.14 due to issue with rpmsg (m4 core) on kernel 4.1.

Tal-eigen, 2018-04-03:
@modjo756 unfortunately no. I will try again in the future though. But currently i have to move on since there is other work that needs my attention. Udoo neo arrived yesterday and i will develop on it now. I will use the IMX6Q for image processing (MT9V034) hopefully the i2c will work :D